openwrt/luci #6117: luci-proto-wireguard status display error.

zdz2019, 2022-11-25:
Pull the latest source code and test that the status interface of luci-proto-wireguard is abnormal.

The overview interface displays abnormally.
[image]

The details interface is displayed normally.
[image]

jow-, 2022-11-25:
This looks like https://github.com/openwrt/luci/commit/90a2b1eaeb58b7169b31fdc097c5bfe6f557c778 is missing. Make sure your `ui.js` is current and maybe clear your browser cache.

zdz2019, 2022-11-25:
you are right thank you!

closed 2022-11-25 by zdz2019