openwrt/luci #6123: ucode exception when accessing overview page

dibdot, 2022-11-30:
Hi,

disclaimer: I haven't followed the latest ucode transitions, so most probably it's a hand made problem on my side.

In my recent private master build I've tried to nuke all lua components, e.g. de-selected "luci-compat" and "luci-lua-runtime" in the LuCI section and various lua-bindings in the Libraries section. After LuCI login I received this error:

[image]

Other LuCI pages are working fine - only the overview page seems to be broken. Is this behavior expected?

Thanks!

```
root@blackhole:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21374-366bcffa0e'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r21374-366bcffa0e'
DISTRIB_TAINTS='no-all'
```

castillofrancodamian, 2022-11-30:
This appears in the [argon-config](https://github.com/jerrykuku/luci-app-argon-config) app.

[image]

root@x86_64:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21374-366bcffa0e'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r21374-366bcffa0e'
DISTRIB_TAINTS='no-all'

jow-, 2022-12-01:
@dibdot - this happens when `ucode-mod-lua` is installed but no `luci-lua-runtime`.

It will trigger this code: 

https://github.com/openwrt/luci/blob/master/modules/luci-mod-status/ucode/template/admin_status/index.ut#L55-L59

... which in turn will attempt to execute a bunch of LuCI libraries. So the solution would be to not include `ucode-mod-lua` in your build.

@castillofrancodamian - this app lacks a dependency on `luci-lib-ipkg`

closed 2022-12-01 by jow-

dibdot, 2022-12-01:
@jow-  many thanks, haven't noticed this dependency - works now!