openwrt/luci #6141: luci-app-pbr: problems with nft domain rules

iainnicol, 2022-12-08:
I appreciate this is bleeding edge stuff! CC @stangri.

I'm running a recent snapshot build, so the packages are fresh off master.

## Steps to reproduce:

1. Open the luci-app-pbr config page
2. Install dnsmasq-full and switch to the "Dnsmasq nft set resolver" resolver set
2. Create a new rule. I set the remote domain to example.org and remote port to 443
3. Save and apply.

## Actual behavior:
There are two errors.

First, something went wrong with the command construction. Possibly address resolution, or string handling. The following command was executed and failed. Note the missing value for daddr:

> nft 'add rule inet fw4 pbr_prerouting  ip daddr  {} tcp dport  {443} goto pbr_mark_0x010000 comment "comment"'

Second, the luci page falls over. The error message cannot be found: `textLabelsTable[element.id]` evaluates to undefined. And then the `.format()` call throws an exception. See https://github.com/openwrt/luci/blob/master/applications/luci-app-pbr/htdocs/luci-static/resources/pbr/status.js#L202

stangri, 2022-12-08:
@iainnicol Please provide additional information as per: https://docs.openwrt.melmac.net/pbr/#getting-help

iainnicol, 2022-12-08:
Thanks. Sorry, haven't spotted that.

Coincidentally, I have completely borked my device so, lol... Once I manage to flash it again, fingers crossed, I'll follow those steps. Now you've shown me I can crank up the verbosity set to 2, I might even be able to figure out the cause myself.

closed 2022-12-08 by iainnicol

iainnicol, 2022-12-08:
So, I'm not quite sure where I went wrong. But after uninstalling pbr (and the luci app), and resetting the config, then reinstalling it's now working grand. Including the resolution of domain names.

It's possible the migration of rules from vpn-policy-routing went pear shaped because my rules were in an invalid state. But, I'm not certain. And, regardless, it's working now.