openwrt/luci #6147: luci-mod-dashboard: wireless information is broken/missing

mistepien, 2022-12-12:
openwrt 22.03.2

in dashboard wireless section there is some missing information. There is no signal bar and no MAC of client. This information is available in default Overview.

DervilRus, 2023-11-07:
> openwrt 22.03.2
> 
> in dashboard wireless section there is some missing information. There is no signal bar and no MAC of client. This information is available in default Overview.

 23.05.0. Xiaomi AX3200 All the same. How did you solve the problem?

stokito, 2023-12-13:
This is a dashboard e.g. a panel for a quick overview.
It's must show only generic inflammation.
You can see details on the Network / Wireless page

http://192.168.1.1/cgi-bin/luci/admin/network/wireless

Here is sources of wifi dashboard for a reference
https://github.com/openwrt/luci/blob/master/modules/luci-mod-dashboard/htdocs/luci-static/resources/view/dashboard/include/30_wifi.js

webysther, 2024-10-07:
> This is a dashboard e.g. a panel for a quick overview. It's must show only generic inflammation. You can see details on the [Network / Wireless](http://192.168.1.1/cgi-bin/luci/admin/network/wireless) page.
> 
> We may add a link on the bottom of Wifi section but I think that's not needed.
> 
> Please add details of close the issue.

I think what he meant is that the dashboard shows the signal column but it is blank and the information is available in the details. If it is not the objective to show the signal column, then it would be better to remove the column, right? Look the purple selection...

[image]

systemcrash, 2024-10-07:
Should look as follows:

[image]

commit 723eb386 (2024-10-07): luci-mod-dashboard: Minor repairs

Strength gauge is now displayed. Appended dBm measurement also.

Corrected some labels.

Closes #6147
commit b42f73fb (2024-10-07): luci-mod-dashboard: Minor repairs

Strength gauge is now displayed. Appended dBm measurement also.

Corrected some labels.

Closes #6147

(cherry picked from commit 723eb386340c8da681aa2c5fb60101690b3231fc)
closed 2024-10-07 by systemcrash