openwrt/luci #6151: luci-xinetd: missing redirect option

lars18th, 2022-12-14:
Hi @he-ma (the maintainer of the package: Helge Mader <ma [at] dev.tdt.de>)

The current implementation of the luci interface for the `xinetd` package doesn't support the `redirect` option. However, the binary service supports this option and it works. I'm using it to redirect some TCP port to a port opened in the loopback by a SSH tunnel. And this works right. So I suggest to _enhance_ the luci interface to support this useful option.

## Steps to reproduce:
Edit your xinetd configuration file `/etc/config/xinetd` and use the `redirect` option for some service like in this example:
```
config service
        option name 'example'
        option flags 'IPv4'
        option type 'UNLISTED'
        option port '8088'
        option protocol 'tcp'
        option socket_type 'stream'
        option user 'nobody'
        option wait 'no'
        option disable 'no'
        option redirect '127.0.0.1 9088'
```

Then check that if you connect to the TCP port 8088 then you will be redirected to the loopback:9088/tcp port.
So the redirection is working, but in the luci xinetd graphical interface the option `redirect` is missing.

## Actual behavior:
The current implemenation doesn't support the `redirect` option _in the LUCI interface_.

## Expected behavior:
It has sense that if the service supports the option, then the luci interface has support for this option.

## Additional Information:
A lot of different solutions are documented to expose an internal TCP port to some external interface. For example, when you're using the simple SSHTUNNEL package to stablish a remote forwarding to the SSH server of a remote client over a NAT connection. In this case in the _server_ (the end of the tunnel) perhaps you want to expose this localhost port to other clients. In this case the most simple (and small) solution is to use the xinetd package with the `redirect` option. However, you can do this _only_ editing the config file, because the luci interface has missing this option.

I hope you'll want to provide support for it.
Regards.

he-ma, 2022-12-14:
@lars18th 
Thank you for your suggestion.
As I am away until January 9th, 2023 I will have a look on this issue if I'm back.
Sorry for the delay.

Regards

lars18th, 2022-12-14:
Hi @he-ma ,

No problem at all. If you will improve this in the next January, then it will nice. Now, I only want to document that the service can handle this and the configuration files support them. Only the LUCI interface requires to add support for this parameter. So don't worry a lot, we can wait. 😉

Regards.

he-ma, 2023-01-11:
@lars18th
Today I sent two PRs upstream to implement the redirect feature you asked for.
https://github.com/openwrt/packages/pull/20264 and https://github.com/openwrt/luci/pull/6194

As one PR is already merged, only the second one has to be merged.

I suggest you close this issue, then.

Regards

closed 2023-01-11 by feckert