openwrt/luci #6153: luci-mod-network: on some platforms, scanning for wireless 2.4G networks does not work since 22.03 RC1

sven3, 2022-12-15:
## Steps to reproduce:
1. Grab one of the following devices: TP-Link TL-WR1043ND v1, TP-Link Archer C7 v5, TP-Link CPE220 v3, Raspberry Pi 3A+, TP-Link TL-WDR3600 v1, D-Link DIR-842 C2, GL.iNet GL-AR750S, GL-MT300N-V2, TP-Link CPE210 v3
2. Login to LuCI
3. Click on "Network" > "Wireless" and click "Scan" on a 2.4G radio

## Actual behavior:
1. No wifi is found, the circle turns and then it says "no data". Firefox' developer console shows `Uncaught (in promise) Error: XHR request timed out at Object.handleReadyStateChange (luci.js?v=git-<version number>)`. Unsure whether this is related.
2. Doing a manual `iw dev wlanX scan` does work and returns wifi networks it found

## Expected behavior:
I would expect LuCI to list the wifi networks, as it worked on all 21.02.x versions

## Additional Information:
This bug was **not** present in OpenWrt 21.02.x. It **is** present since at least 22.03 RC1. It has been confirmed to exist by several users in all subsequent RCs and in all final 22.03.x versions. See here: https://github.com/openwrt/openwrt/issues/10158
It seems that mostly ath79 devices are affected by this bug, but the bug seems not limited to ath79.

jow-, 2022-12-15:
Can't reproduce on a Buffalo WZR-600DHP with a fresh flash of 22.03.2

sven3, 2022-12-15:
I can provide you with SSH access to a TP-Link TL-WR1043ND v1, if this helps you. You could tunnel LuCI over SSH.

sven3, 2022-12-15:
I just tried today on a Trendnet TEW-692GR (can't reproduce, works like a charm) and on a TP-Link TL-WR1043ND v1 (problem is still present). Both on 22.03.2.

jow-, 2022-12-15:
Problem only occurs when scanning through wpa_supplicant (or hostapd?) on an ujail enabled device. So commonly on tiny devices which are already joined to a network.

Workaround: `mv /sbin/ujail /sbin/ujail.hide && reboot`
Bugfix: https://github.com/openwrt/openwrt/pull/11280/files#diff-b2de369571d0d39ce8545fa2d6b1b6021ff4c101f73498570bc4da702832202b

Please follow up on https://github.com/openwrt/openwrt/pull/11280, not a LuCI issue.

closed 2022-12-15 by jow-

sven3, 2022-12-15:
Wow, that was fast. Many thanks!!

humaita-github, 2022-12-15:
Hi, thank you for the quick workaround. `mv` will not work on tiny devices, as it seems to create of new copy of ujail and there might not be enough free space. 

Since your workaround basically incapacitates ujail, we can just directly remove it: `rm /sbin/ujail && reboot`

This worked on a D-Link DIR-505.