openwrt/luci #6156: luci-app-opkg: Changing to next page in "Installed" tab does not work

minax007, 2022-12-17:
## Steps to reproduce:

1. go to: System → Software → Installed 
2. Scroll down the package until you reach the end
3. Click on the "next page arrows" on the right corner

## Actual behavior:

You can click on the arrows on the BOTTOM right corner but it does not work. 
If you scroll up and click on the arrows on the TOP right corner ... that works. 
After having clicked the arrows on the TOP right corner also the arrows on the BOTTOM right corner work. 

## Expected behavior:

That it switches to the next page for illustrating further packages. 

```
If you change to the "Available" tab (instead of "Installed" ... there it works as expected. 
```

hnyman, 2022-12-17:
Sounds like a duplicate of #6007 

And it seems to work ok for me, with the current master.
LuCI Master (git-22.347.45170-440a2e7) / OpenWrt SNAPSHOT (r21516-1b8c8864c5)

minax007, 2022-12-17:
@hnyman 

Thanks for your response!

I am using 
OpenWrt SNAPSHOT r21477-90120a1552 with LuCI git-22.297.83017-0143ef2
Browser: Firefox 107.0.1 (64-Bit)

DragonBluep, 2023-01-08:
@hnyman This problem still exists, and it is triggered only under specific circumstances.

Steps to reproduce:
Close the browser and reopen it. Then log in to luCI and enter the opkg installed page. Do not press any other button. Then you will find that the next page button **at the bottom of the page** does not work.
Once you click the next page button **at the top of the page**, the next page button **at the bottom of the page** will be activated.

hnyman, 2023-01-09:
cc @jow- 

LuCI Master (git-22.361.69865-deed682) / OpenWrt SNAPSHOT (r21729-89eb8b50d1)

I just tested with Microsoft Edge Version 108.0.1462.76 (Official build) (64-bit)  and Firefox 109.0b9 (64-bit)   
and it works...
...but not ok. It changes page wrongly.

I have 314 packages installed. (All are included in the firmware, none have been installed later). 

The Software/installed page first shows packages 1-100. 

* If I push **the button at the bottom**, it jumps over the second page and shows packages 201-300.   
  If I then use back button, it goes to 101-200, and after that it pages normally back/forward.

* If I push **the top button**, it shows packages 101-200 as expected.

That happens quite consistently on both routers.

If you have installed less than 200 packages (or less than 300?), it probably does not want to jump to the "non-existent" page and that's why it seems to do nothing.

hnyman, 2023-01-09:
The same actually also happens with the "Available" page:  
the bottom button leads to page 3 with packages 201-300 instead of 101-200.

jow-, 2023-01-09:
Fixed by ec3aac47c43d44d170af6a09d31493c2e8efe590

closed 2023-01-09 by jow-