openwrt/luci #6162: luci-base: proto autoip shows "Unsupported protocol type."

mibofra, 2022-12-19:
## Steps to reproduce:
https://github.com/openwrt/packages/blob/master/libs/avahi/files/netifd-autoip.sh → avahi-autoipd provides `autoip` protocol

## Actual behavior:
Surely, the protocol works, but Luci does not recognize the protocol. This can lead an user using Luci to think that even if it is working, it is not.
And, surely, there are not the statistics of that interface in the network section, for this reason.

## Expected behavior:
Luci should make the possibility to use `autoip` protocol under the possible protocols for an interface, if avahi-autoipd package is installed.

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''
```

`autoipd` provides IPv4 Local Link addresses, that can be useful in some cases, as an example, meshing.
Obviously, I do not want to discute if meshing via v4 is a devilish thing or not, but users can use autoipd for whatever reasons.

I am simply asking because it behaves like `unmanaged` protocol, it should be simple to implement.
But as it is UCI supported, using `unmanaged` instead force to do something else to manage avahi-autoipd to start and stop on the interface, making the existence of the `autoip` protocol useful only in cli.

Thanks in advance for any answers :)

jow-, 2022-12-19:
Please psot the working uci configuration here.

closed 2022-12-19 by jow-

jow-, 2022-12-19:
You will need to install/include `luci-proto-autoip` once it becomes available.

mibofra, 2022-12-19:
@jow- already understood seeing the commit, but I wanted to thank you anyway :)

commit e112bde5 (2022-12-19): luci-proto-autoip: add protocol support for avahi-autoipd

Fixes: #6162

(cherry picked from commit 2965e527f578abda1dc58dd540343422455be3ac)