openwrt/luci #6168: luci-app-dockerman: sub menu not showing

ysc3839, 2022-12-23:
## Steps to reproduce:
1. Install `luci-app-dockerman`.
2. Reload luci page.

## Actual behavior:
The `Docker` menu contains no sub menu.
[image]

## Expected behavior:
The `Docker` menu should contains sub menu.
[image]

## Additional Information:
Replace `alias("admin", "docker", "config")` with `firstchild()` resolve this issue.
```diff
--- a/applications/luci-app-dockerman/luasrc/controller/dockerman.lua
+++ b/applications/luci-app-dockerman/luasrc/controller/dockerman.lua
@@ -9,7 +9,7 @@ module("luci.controller.dockerman",package.seeall)

 function index()
        entry({"admin", "docker"},
-               alias("admin", "docker", "config"),
+               firstchild(),
                _("Docker"),
                40).acl_depends = { "luci-app-dockerman" }
```
IIRC there's no such issue in previous version of luci. Is this expected behavior?

OpenWrt version information from system `/etc/openwrt_release`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''
```

dannygonzalez0861, 2022-12-28:
I just updated OPKG and the same issue has now happened to me. My dockers are running as i can still access them but I too have lost the sub menu's as yuo have:
[image]

JKSTAFF, 2023-01-05:
I make a snapshot image with dockerman integrated. Droplist menu lost after reconfig docker root path or create some containers. Luci unable to dispatch.

kofec, 2023-01-05:
I confirm the same issue on latest 22.03 but in 21.02 all is OK

kofec, 2023-01-05:
Hi @lisaac https://github.com/lisaac/luci-app-dockerman - maybe we should report here ?

kofec, 2023-01-05:
@ysc3839 Thanks - your workaround/solution is working fine

hnyman, 2023-01-14:
luci-app-dockerman in master was apparently broken and then fixed in April with commit 0c1fc7f6 from PR #5789
Package in 22.03 was fine at that time.
22.03 has then got broken in Dec 2022 due to backport commit fc759abea 

I will backport the dockerman fix 0c1fc7f6  from master.

hnyman, 2023-01-14:
Shoudl be fixed with 68e23b284

commit 68e23b28 (2023-01-14): luci-app-dockerman:  Update to play nice with recent changes

https://github.com/openwrt/luci/commit/180d39dcd2427e3c32c0ec7ecc3c7bfb48c0d0ab broke dockerman's appearance in the menus

Make a small change to dockerman's index entry to make it play nice with the new architecture, firstchild() was obtained using DAWN's controller/dawn.lua as an example

(cherry picked from commit 0c1fc7f64566bd5da643eadc6b817bb0f5f591e7)
Fixes #6168
closed 2023-01-14 by hnyman