openwrt/luci #6169: luci-nginx: running as non-root user?

ysc3839, 2022-12-23:
## Steps to reproduce:
1. Install `luci-nginx`.
2. Replace `user root;` with `user ubus;` in `/etc/nginx/uci.conf.template`.
3. `/etc/init.d/nginx restart`
4. Reload luci page.

## Actual behavior:
The page always shows `Session expired`.

## Expected behavior:
luci works fine.

## Additional Information:
In DevTools, `/ubus` requests always returns `Access denied`:
```
[{"jsonrpc":"2.0","id":null,"error":{"code":-32002,"message":"Access denied"}}]
```
In my opinion, anyone who can access ubus socket should be able to make rpc requests.

OpenWrt version information from system `/etc/openwrt_release`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''
```

jow-, 2022-12-25:
The ubus daemon disallows access by non-root users by default, regardless of the socket file permissions. You need to to place ACL rules for non-uid-0 users selectively granting procedure access. See `/usr/share/acl.d/` (*not* `/usr/share/rpcd/acl.d/`) for examples. The former path belongs to OepnWrt/ubusd itself and is outside of the scope of LuCI.

closed 2022-12-25 by jow-

ysc3839, 2022-12-26:
@jow- Thanks. By the way, what `access` should I allow in acl to make luci work properly?

jow-, 2022-12-26:
I think a definition like the following should do the trick:

```json
{
  "user": "ubus",
  "access": {
    "*": [ "*" ]
  }
}
```

The above ACL will allow access to any method. You could also try to limit it somewhat by replicating LuCI's rpcd ACL rules (see the "ubus" scope `ubus call session list` on a system where you're currently logged into LuCI).

```json
{
  "user": "ubus",
  "access": {
    "dsl": [
      "metrics"
    ],
    "file": [
      "read",
      "write",
      "list",
      "remove",
      "exec",
      "stat"
    ],
    "hostapd.*": [
      "del_client",
      "wps_start",
      "wps_cancel",
      "wps_status"
    ],
    "ieee1905": [
      "get_topology"
    ],
    "iwinfo": [
      "assoclist",
      "countrylist",
      "freqlist",
      "txpowerlist",
      "scan",
      "info"
    ],
    "luci": [
      "setInitAction",
      "getConntrackHelpers",
      "getMountPoints",
      "getFeatures",
      "getSwconfigFeatures",
      "getSwconfigPortState",
      "getVersion",
      "getConntrackList",
      "getRealtimeStats",
      "getProcessList",
      "getInitList",
      "getLEDs",
      "getTimezones",
      "getUSBDevices",
      "setLocaltime",
      "setPassword",
      "getBlockDevices",
      "setBlockDetect"
    ],
    "luci-rpc": [
      "getHostHints",
      "getBoardJSON",
      "getNetworkDevices",
      "getWirelessDevices",
      "getDHCPLeases",
      "getDUIDHints"
    ],
    "luci.ddns": [
      "get_services_status",
      "get_ddns_state",
      "get_env",
      "get_services_log"
    ],
    "luci.upnp": [
      "get_status",
      "delete_rule"
    ],
    "luci.wireguard": [
      "getWgInstances",
      "generateKeyPair",
      "getPublicAndPrivateKeyFromPrivate",
      "generatePsk"
    ],
    "network": [
      "get_proto_handlers"
    ],
    "network.interface": [
      "dump"
    ],
    "network.rrdns": [
      "lookup"
    ],
    "session": [
      "access",
      "login"
    ],
    "system": [
      "board",
      "info",
      "validate_firmware_image",
      "reboot"
    ],
    "uci": [
      "changes",
      "get",
      "add",
      "apply",
      "confirm",
      "delete",
      "order",
      "rename",
      "set"
    ]
  }
}
```

ysc3839, 2022-12-27:
@jow- Thank you very much!