openwrt/luci #6172: luci_statistics RRDTool error: There is no RRD data available yet to render graphs.

IvoKvesic, 2022-12-26:
***

## Steps to reproduce:
1. Go to Statistics/Graphs

## Actual behavior:
The following error message appears:

_There is no RRD data available yet to render graphs.
You need to configure collectd to gather data into .rrd files._

## Expected behavior:
Expect to see the graph.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`:

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''

Contents of the collectd.conf:

_BaseDir "/var/run/collectd"
Include "/etc/collectd/conf.d/*.conf"
PIDFile "/var/run/collectd.pid"
PluginDir "/usr/lib/collectd"
TypesDB "/usr/share/collectd/types.db"
Interval 30
ReadThreads 2

LoadPlugin ipstatistics

LoadPlugin rrdtool
<Plugin rrdtool>
	DataDir "/etc/rrd"
	StepSize 30
	HeartBeat 60
	RRARows 288
	RRASingle true
	RRATimespan 3600
	RRATimespan 86400
	RRATimespan 2678400
</Plugin>

LoadPlugin thermal_

hnyman, 2022-12-27:
Luci statistics in general works just fine, so this is likely something related to your config.

First guess is that you have made some changes to Luci statistics config (which you didn't show), so that collectd doesn't like them when converted to native collectd settings.

You have at least changed the datadir from /tmp ramdisk to /etc/rrd on flash (which causes flash wear).
> DataDir "/etc/rrd"
Hopefully the directory permissions have been adjusted right.

The error message hosts toward config file incompatibility, so likely some plugin config is incomplete, or some general setting is missing.

Are there other changes?  Ipstatistics plugin has apparently been enabled, but is something disabled?
You might test plugin by plugin, which causes trouble for you.

IvoKvesic, 2022-12-27:
Thank you very much for your kind answer. I have done just the default installation without changing any settings and I was getting the above mentioned error immediately. Then I noticed that the /tmp/rrd was not existing so I created it manually (with world access rights). This didn't help. Then I tried changing the data directory to /opt/rrd (opt being on an USB drive which I use for stuff that writes a lot of data like e.g. Yamon). No change, still the same error. Then I tried the /etc/rrd that you noticed in my OP. 

These moves with the data dir (always with full access rights) were the only items that I played with. The only general plugins I have is thermal and sensors. RRDTool is the only output plugin which is enabled. Should I add anything else? The only thing I want to see is the temperatures.

hnyman, 2022-12-27:
> I have done just the default installation without changing any settings and I was getting the above mentioned error immediately. 

...

> LoadPlugin ipstatistics

Except that you have enabled e.g. the non-default ipstatistics plugin. So you have made other changes...

Likely you have now a somehow dirty /etc/config/luci_statistics

I suggest that you copy a clean copy from /rom/etc/config (is LuCI statistics was installed in the flashed image), or you use opkg to reinstall luci-app-statistics, and then use the /etc/config/luci_statistics-opkg as the new clean /etc/config/lucistatistics.   The default data dir /tmp/rrd is created automatically if missing.

Make also sure that you have installed the needed collectd-mod-... for the statis items that you enable.

> The only general plugins I have is thermal and sensors. 

Note that the sensors plugin can be tricky as the available sensors vary. You might need to first configure sensors via LuCI config.

IvoKvesic, 2022-12-27:
Thanks a lot again for your continued kind support. I have now carefully removed everything related to luci statistics, including manual removal of files and directories left all over the place. Then I re-installed luci-app-statistics and only the thermal and sensors collectd-mods. Everything works fine now, so obviously in my previous installation I must have screwed up something, but I really can't find out what.  I have also found [this nice stuff](https://github.com/gSpotx2f/luci-app-temp-status) that adds real-time temperatures to the status overview page.

alitlyyee, 2022-12-27:
>

hnyman, 2022-12-27:
Great that it works ok for you now.

closed 2022-12-27 by hnyman