openwrt/luci #6177: Invisible buttons due to wrong font color

JJWatMyself, 2022-12-30:
> I cannot reproduce this error. I have deleted the wan/wan6 interface. (or removed the switch port "wan" from the wan/wan6 interface) I then added the switchport "wan" to the lan bridge. VLAN tagging also works. The only thing I'm noticing right now is that with Luci there is problem to set the IP address from static to dhcp on br-lan.

Notes from things I worked on last night are below:

My 'things going horribly wrong' comment on Luci was a simple static IP address change. So might not just be change from static to DHCP.  More exploring on that might be good.

I do notice that the IP address change warning dialog has no visible clickable button though. The colors for two invisible buttons ('Cancel' button.btn and 'Apply with revert after connectivity loss' button.btn.cbi-button-action.im) shows as #FFFFF instead of #212322.  That is theme OpenWRT2020, whereas the BootStrap themes seems to be fine.

Re can't reproduce the 'network loop' issue I continue to see when adding WAN, this is an interesting one. Can you paste your /etc/config/network so I can test an identical config as well as diff the config where I can reproduce?  Also confirm you are running the same code you linked for me to test please.

_Originally posted by @JJWatMyself in https://github.com/openwrt/openwrt/issues/11579#issuecomment-1362910795_

JJWatMyself, 2022-12-30:
I think the issue is in this file:

[https://github.com/openwrt/luci/blob/master/themes/luci-theme-openwrt/htdocs/luci-static/openwrt.org/cascade.css](https://github.com/openwrt/luci/blob/master/themes/luci-theme-openwrt/htdocs/luci-static/openwrt.org/cascade.css)

Potentially line 723 (Maybe also 729, 734) is where I think the white on white issue exists.  I think it should be #212322 instead?

`color: #fff;`

```

.cbi-button-action.important,
.cbi-page-actions .cbi-button-apply,
.cbi-section-actions .cbi-button-edit {
	color: #fff;
	background: #37c;
}

.cbi-button-positive.important,
.cbi-page-actions .cbi-button-save {
	color: #fff;
	background: #595;
}

.cbi-button-negative.important {
	color: #fff;
	background: #a22;
}

.cbi-page-actions .cbi-button-apply + .cbi-button-save,
.cbi-page-actions .cbi-button-negative + .cbi-button-save {
	background: #fff;
	color: #595;
}
```

I also am not sure where the color for button.btn is but it also had the same issue.

Reproduced by changing IP address and the warning screen that follows.

feckert, 2023-01-09:
Can you please take pictures and attach them? Otherwise I don't think anyone can help you figure out what the problem is.

JJWatMyself, 2023-01-09:
@feckert I think I have figured out the problem and is in the comment above you post.

feckert, 2023-01-09:
Is this now a problem, that we should solve? If **yes** then please open a pullrequest and if **not** then please close this issue.

commit 1597d8e5 (2023-02-07): luci-theme-openwrt-2020: don't inherit background in alert message buttons

Fixes: #6177
commit 9d948051 (2023-02-07): luci-theme-openwrt-2020: don't inherit background in alert message buttons

Fixes: #6177

(cherry picked from commit 1597d8e555cfe7444bcbcb44d45caf6954b61b3e)
closed 2023-02-07 by jow-