openwrt/luci #6178: <luci-proto-sstp> Custom Port

aminsamizadeh, 2022-12-31:
the issue is not being able to have custom port for servers
per having discussion with the creator of the sstp-client custom port can be used by cli but for having custom port with Luci , it has to be implemented to have in the "ssto server" like for example "10.10.10.10:9443(custom port)" other than 443 port.

feckert, 2023-01-09:
To be ably to make this change easy in LuCI. Please extend the [netifd proto handler](https://github.com/openwrt/packages/blob/master/net/sstp-client/files/lib/netifd/proto/sstp.sh) with a new uci option called **port**.
If the new uci **port** option is set, then add this to the [server variable](https://github.com/openwrt/packages/blob/master/net/sstp-client/files/lib/netifd/proto/sstp.sh#L85) during setup call.

dannil, 2024-02-29:
This got fixed by #6634, I vote for closing this issue.

systemcrash, 2024-03-01:
Resolved by #6634

closed 2024-03-01 by systemcrash