openwrt/luci #6180: luci-mod-network: IPsets configuration under dnsmasq appears 2 times

ajayramaswamy, 2023-01-02:
## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. openwrt master snapshot of 1/1/2023
2. luci Powered by LuCI Master (git-22.361.69865-deed682)
3. login - select network - dhcp and dns

## Actual behavior:
see attached screenshot
[image]

## Expected behavior:
only the new tab on the top should be active the redundant section in the middle should be removed

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

**Example: Replace the following lines and remove this**
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21614+102-425945c313'
DISTRIB_TARGET='rockchip/armv8'
DISTRIB_ARCH='aarch64_generic'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT 2023-01-01-slim'
DISTRIB_TAINTS='no-all busybox'
```

jow-, 2023-02-07:
It's not that simple actually. The IP Sets tab actually creates ipsets only while the IP sets option has code for automatically translating into nftsets in case nft/fw4 is used.

There currently is a lack of a dedicated "nftset" section support. I'm inclined to drop/remove the ipsets tab in favor to keeping the ip set option since it has better compatibility.

dave14305, 2023-02-18:
Isn’t the old ipset option obsoleted in this commit, so the only supported option now in master is the `ipset` section (and therefore the IPsets tab)?

https://github.com/openwrt/openwrt/commit/d7f378796f985c902ff7906767c275c40a15347f