openwrt/luci #6195: rpcd-mod-luci: error: 'IWINFO_CIPHER_CCMP256' undeclared

DocMAX, 2023-01-11:
on fresh openwrt snapshot:
```
-fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/docmax/openwrt/fb7530/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/rpcd-mod-luci-20210614=rpcd-mod-luci-20210614 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/docmax/openwrt/fb7530/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libnl-tiny -I/home/docmax/openwrt/fb7530/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -DNDEBUG -fPIC   -Os -Wall -Werror --std=gnu99 -g3 -Wmissing-declarations -MD -MT CMakeFiles/rpcd-mod-luci.dir/luci.c.o -MF CMakeFiles/rpcd-mod-luci.dir/luci.c.o.d -o CMakeFiles/rpcd-mod-luci.dir/luci.c.o -c /home/docmax/openwrt/fb7530/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/rpcd-mod-luci-20210614/luci.c
/home/docmax/openwrt/fb7530/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/rpcd-mod-luci-20210614/luci.c: In function 'rpc_luci_get_iwinfo':
/home/docmax/openwrt/fb7530/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/rpcd-mod-luci-20210614/luci.c:1086:44: error: 'IWINFO_CIPHER_CCMP256' undeclared (first use in this function); did you mean 'IWINFO_CIPHER_CCMP'?
 1086 |                                 if (nret & IWINFO_CIPHER_CCMP256)
      |                                            ^~~~~~~~~~~~~~~~~~~~~
      |                                            IWINFO_CIPHER_CCMP
/home/docmax/openwrt/fb7530/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/rpcd-mod-luci-20210614/luci.c:1086:44: note: each undeclared identifier is reported only once for each function it appears in
/home/docmax/openwrt/fb7530/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/rpcd-mod-luci-20210614/luci.c:1092:44: error: 'IWINFO_CIPHER_GCMP256' undeclared (first use in this function); did you mean 'IWINFO_CIPHER_GCMP'?
 1092 |                                 if (nret & IWINFO_CIPHER_GCMP256)
      |                                            ^~~~~~~~~~~~~~~~~~~~~
      |                                            IWINFO_CIPHER_GCMP
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:57: /home/docmax/openwrt/fb7530/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/rpcd-mod-luci-20210614/.built] Error 1
make[2]: Leaving directory '/home/docmax/openwrt/fb7530/feeds/luci/libs/rpcd-mod-luci'
time: package/feeds/luci/rpcd-mod-luci/compile#0.11#0.01#0.12
    ERROR: package/feeds/luci/rpcd-mod-luci failed to build.
make[1]: *** [package/Makefile:116: package/feeds/luci/rpcd-mod-luci/compile] Error 1
make[1]: Leaving directory '/home/docmax/openwrt/fb7530'
make: *** [/home/docmax/openwrt/fb7530/include/toplevel.mk:230: package/feeds/luci/rpcd-mod-luci/compile] Error 2
```

jow-, 2023-01-11:
What's the output of `./scritps/getver.sh` in your buildroot?

DocMAX, 2023-01-11:
r20082-b68e9f2e19

jow-, 2023-01-11:
That version is from June 22th, the iwinfo bump introducing the 256bit ciphers was done in August with https://github.com/openwrt/openwrt/commit/cc6a323e2328176b732b13f1f09745354270cd39 - your buildroot is too old, this does not appear to be a fresh snapshot.

DocMAX, 2023-01-11:
ok sorry, indeed it's a fork for fritzbox 7530 dsl support. probably my fault. closing.

closed 2023-01-11 by DocMAX