openwrt/luci #6198: luci-app-pbr: recursive dependency

feckert, 2023-01-16:
## Steps to reproduce:
@stangri During build tree preparation I get the following error for package "luci-app-pbr"
```
tmp/.config-package.in:57264:error: recursive dependency detected!
tmp/.config-package.in:57264:   symbol PACKAGE_luci-app-pbr depends on PACKAGE_luci-app-pbr
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
```
I think the problem is the [pbr package](https://github.com/openwrt/packages/blob/master/net/pbr/Makefile) in the package feed.

This is the generate section of the .config-package.in file
```
config PACKAGE_luci-app-pbr
    tristate "luci-app-pbr......................... Policy Based Routing Service Web UI"
    default y if DEFAULT_luci-app-pbr
    default m if ALL
    select PACKAGE_jsonfilter
    select PACKAGE_pbr if PACKAGE_pbr-iptables<PACKAGE_luci-app-pbr
    select PACKAGE_libc
    depends on !(PACKAGE_pbr-iptables<PACKAGE_luci-app-pbr) || IPV6
    depends on !(PACKAGE_pbr-iptables<PACKAGE_luci-app-pbr && IPV6) || IPV6
    select PACKAGE_luci-base
    help
      Provides Web UI for Policy Based Routing Service.
      https://github.com/openwrt/luci
      OpenWrt LuCI community

```

I think the problem is the PROVIDE line https://github.com/openwrt/packages/blob/master/net/pbr/Makefile#L37

@stangri can you please have a look on this issue?

## Additional Information:
Latest openwrt master

stangri, 2023-01-16:
@feckert do you have a suggestion on how can I change the Makefile so that variants still provide `pbr` but it doesn't cause the recursive dependency?

feckert, 2023-01-17:
@stangri thanks your pullrequest fixed it. No more error messages.

closed 2023-01-17 by feckert