openwrt/luci #6200: luci-app-ddns: List of DDNS Service provider is unordered

bdrung, 2023-01-17:
## Steps to reproduce:

1. go to: Services → Dynamic DNS → Add new services...
2. click on the dropdown menu for "DDNS Service provider"

## Actual behavior:

The list of DDNS Service provider is unordered.

## Expected behavior:

The list of DDNS Service provider should be ordered alphabetical.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.3'
DISTRIB_REVISION='r20028-43d71ad93e'
DISTRIB_TARGET='ipq806x/generic'
DISTRIB_ARCH='arm_cortex-a15_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 22.03.3 r20028-43d71ad93e'
DISTRIB_TAINTS=''
```

systemcrash, 2023-01-18:
Looks in order to me - can you trim a screenshot down to the problem area?

bdrung, 2023-01-18:
[image]

jow-, 2023-01-18:
@feckert yes but the dropdowns are populated by traversing the `this._services` dict.

Here's my attempt to solve this, maybe someone can test:

```
diff --git a/applications/luci-app-ddns/htdocs/luci-static/resources/view/ddns/overview.js b/applications/luci-app-ddns/htdocs/luci-static/resources/view/ddns/overview.js
index 4592c47c4c..afba9641c7 100644
--- a/applications/luci-app-ddns/htdocs/luci-static/resources/view/ddns/overview.js
+++ b/applications/luci-app-ddns/htdocs/luci-static/resources/view/ddns/overview.js
@@ -515,8 +515,7 @@ return view.extend({
                        service_name = s2.option(form.ListValue, 'service_name',
                                        String.format('%s', _("DDNS Service provider")));
                        service_name.value('-',"-- " + _("custom") + " --");
-                       for (var elem in _this.services)
-                               service_name.value(elem);
+                       Object.keys(_this.services).sort().forEach(name => service_name.value(name));
                        service_name.validate = function(section_id, value) {
                                if (value == '') return _("Select a service");
                                if (s2.service_supported == null) return _("Checking the service support...");
@@ -638,8 +637,7 @@ return view.extend({
                                        String.format('%s', _("DDNS Service provider")));
                                service_name.modalonly = true;
                                service_name.value('-',"-- " + _("custom") + " --");
-                               for (var elem in _this.services)
-                                       service_name.value(elem);
+                               Object.keys(_this.services).sort().forEach(name => service_name.value(name));
                                service_name.cfgvalue = function(section_id) {
                                        return uci.get('ddns', section_id, 'service_name') || '-';
                                };
```

bdrung, 2023-01-18:
Tested this diff and it is working.

commit d9dcf987 (2023-04-28): luci-app-ddns: order list of DDNS service providers

Fixes: #6200
commit 4390b4dc (2023-04-28): luci-app-ddns: order list of DDNS service providers

Fixes: #6200

(cherry picked from commit d9dcf98778f46f51d56d3df2a19a4245a0903b7c)
closed 2023-04-28 by jow-