openwrt/luci #6210: luci-mod-dashboard: wireless data statistics overflow when it exceeds 2GB

DragonBluep, 2023-01-23:
## Steps to reproduce:
Download something larger than 2GB.

## Actual behavior:
The tx data statistics becomes a negative number when it exceeds 2GB. Once it exceeds 2GB again, it will back to 0.

[image]

## Additional Information:

**MT7620 is a 32-bit processor. I guess it should be related to this. Since I am not familiar with js, I cannot provide more information.**

OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOTS'
DISTRIB_REVISION='Custom build based on master branch 2023/01/22'
DISTRIB_TARGET='ramips/mt7620'
DISTRIB_ARCH='mipsel'
```

st7105, 2023-02-25:
Actual
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r22135-530f5c2fda'
DISTRIB_TARGET='ipq807x/generic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r22135-530f5c2fda'
DISTRIB_TAINTS=''
```

t-8ch, 2023-06-08:
These patches should fix it:

https://lists.openwrt.org/pipermail/openwrt-devel/2023-May/041056.html
https://lists.openwrt.org/pipermail/openwrt-devel/2023-May/041096.html

DragonBluep, 2023-07-03:
Fixed by https://github.com/openwrt/openwrt/commit/a226153067766db6d733aa3e330742c9da40c4f6 and https://github.com/openwrt/openwrt/commit/2486841c1bc7c3267e409b510e142e04cc5f0a5b

closed 2023-07-03 by DragonBluep