openwrt/luci #6229: interfaces: Editing virtual dynamic interface throws "Cannot read properties of null (reading '.name')"

McGiverGim, 2023-02-08:
## Steps to reproduce:

I have a PPPoE ISP connection, with both, IPv4 and IPv6. The IPv6 appears in luci as a "Virtual dynamic interface" created automatically. 
1. go to: Network → Interfaces and search for the Virtual Dynamic Interface created.
2. Press the edit button
3. A "Cannot read properties of null (reading '.name')" appears on top.

## Actual behavior:
An error appears

## Expected behavior:
I don't know if there is something that can be edited there. So I don't know what is the expected behavior:
- Don't throw the error, and show an edit window, or
- Don't let edit the virtual dynamic interface, letting the button disabled.

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21968-acd8e94d20'
DISTRIB_TARGET='ipq807x/generic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r21968-acd8e94d20'
DISTRIB_TAINTS=''
```

raywang12, 2023-04-20:
I also faced this problem.
For the "virtual dynamic interface" we can see  the following explanation in netwrok.js
	/**
	 * Checks whether this logical interface is dynamic.
	 *
	 * A dynamic interface is an interface which has been created at runtime,
	 * e.g. as sub-interface of another interface, but which is not backed by
	 * any user configuration. Such dynamic interfaces cannot be edited but
	 * only brought down or restarted.
	 *
	 * @returns {boolean}
	 * Returns a boolean indicating whether this interface is dynamic (`true`)
	 * or not (`false`).
	 */

But in the interface.js the restart and stop buttons disabled
As  explanation in netwrok.js I think the edit and delete button should be disabled?

blacknell, 2024-03-15:
I'm seeing the same error. I get 
```log
TypeError
Cannot read properties of null (reading '.name')
```

Also, clicking delete for this interface (I've seen this done on tutorials) but it has no effect.