openwrt/luci #6230: luci: stuck at "Loading view..."

arinc9, 2023-02-08:
## Steps to reproduce:
1. install luci-light package

## Actual behavior:
The page is stuck at "Loading view…"

## Expected behavior:
The login page to appear.

## Additional Information:
Installed LuCI packages
```
luci-app-attendedsysupgrade - git-22.285.67526-18bfcca
luci-app-firewall - git-23.035.45612-c54efde
luci-app-opkg - git-23.009.82915-ec3aac4
luci-base - git-23.038.47117-2234d0c
luci-light - git-23.024.33244-34dee82
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-23.038.47117-2234d0c
luci-mod-status - git-23.038.33206-b8f91d5
luci-mod-system - git-23.013.73089-0186d7e
luci-proto-ipv6 - git-21.148.48881-79947af
luci-proto-ppp - git-21.158.38888-88b9d84
luci-theme-bootstrap - git-23.038.32982-f416c27
```

Console log
```
GEThttp://192.168.1.1/cgi-bin/luci/
[HTTP/1.1 403 Forbidden 63ms]

This page uses the non standard property “zoom”. Consider using calc() in the relevant property values, or using “transform” along with “transform-origin: 0 0”. luci
Password fields present on an insecure (http://) page. This is a security risk that allows user login credentials to be stolen. luci
SyntaxError: missing ) after condition
  in http://192.168.1.1/luci-static/resources/ui.js?v=git-23.038.47117-2234d0c:25
  at require/compileClass/< (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:179:29)
  at promise callback*compileClass (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:178:29)
  at promise callback*require (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:184:112)
  at __init__ (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:151:282)
  at ClassConstructor (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:11:78)
  at  (http://192.168.1.1/cgi-bin/luci/:26:6)
    raise http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:155
    compileClass http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:179
    promise callback*compileClass http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:178
    promise callback*require http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:184
    __init__ http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:151
    ClassConstructor http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:11
    <anonymous> http://192.168.1.1/cgi-bin/luci/:26
luci.js:161:9
SyntaxError: missing ) after condition
  in http://192.168.1.1/luci-static/resources/ui.js?v=git-23.038.47117-2234d0c:25
  at require/compileClass/< (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:179:29)
  at promise callback*compileClass (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:178:29)
  at promise callback*require (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:184:112)
  at __init__ (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:151:282)
  at ClassConstructor (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:11:78)
  at  (http://192.168.1.1/cgi-bin/luci/:26:6)
    raise http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:155
    compileClass http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:179
    promise callback*compileClass http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:178
    promise callback*require http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:184
    __init__ http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:151
    ClassConstructor http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:11
    <anonymous> http://192.168.1.1/cgi-bin/luci/:26
luci.js:161:9
Uncaught (in promise) SyntaxError: missing ) after condition
  in http://192.168.1.1/luci-static/resources/ui.js?v=git-23.038.47117-2234d0c:25
  at require/compileClass/< (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:179:29)
  at promise callback*compileClass (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:178:29)
  at promise callback*require (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:184:112)
  at __init__ (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:151:282)
  at ClassConstructor (http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:11:78)
  at  (http://192.168.1.1/cgi-bin/luci/:26:6)
    raise http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:155
    compileClass http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:179
    promise callback*compileClass http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:178
    promise callback*require http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:184
    __init__ http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:151
    ClassConstructor http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:11
    <anonymous> http://192.168.1.1/cgi-bin/luci/:26
2 luci.js:155:3
Uncaught (in promise) TypeError: classes.ui.addNotification is not a function
    error http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:163
    promise callback*__init__ http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:151
    ClassConstructor http://192.168.1.1/luci-static/resources/luci.js?v=git-23.038.47117-2234d0c:11
    <anonymous> http://192.168.1.1/cgi-bin/luci/:26
luci.js:163:12
```

OpenWrt version information from system `/etc/openwrt_release`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r22014-628870146d'
DISTRIB_TARGET='bcm53xx/generic'
DISTRIB_ARCH='arm_cortex-a9'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r22014-628870146d'
DISTRIB_TAINTS='no-all'
```

jow-, 2023-02-08:
Already fixed by https://github.com/openwrt/luci/commit/41e9b8d1219a949a0a3105eabec3774dc97f65c2

closed 2023-02-08 by jow-

arinc9, 2023-02-08:
Hmm I just cleaned and refreshed the feeds. git log on ./feeds/luci shows `Merge pull request #6228 from weblate/weblate-openwrt-luci` as the latest commit but still getting stuck. I don't know how to cross reference the versioning on these packages with this repository so I'm not sure if the fix is actually included or not.

```
luci-app-firewall - git-23.035.45612-c54efde
luci-base - git-23.039.28596-41e9b8d
luci-light - git-23.024.33244-34dee82
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-23.038.47117-2234d0c
luci-mod-status - git-23.038.33206-b8f91d5
luci-mod-system - git-23.013.73089-0186d7e
luci-proto-ipv6 - git-21.148.48881-79947af
luci-proto-ppp - git-21.158.38888-88b9d84
luci-ssl - git-23.035.26083-7550ad6
luci-theme-bootstrap - git-23.038.32982-f416c27
```

jow-, 2023-02-08:
Make sure to clear your browser cache.

arinc9, 2023-02-08:
Turns out the OpenWrt SDK was just refusing to add the newest compiled package into the image. It started working after a make distclean. Thanks Jo.

jow-, 2023-02-08:
Thanks for confirming @arinc9

jow-, 2023-02-09:
Quickfix: 

```
cp /rom/www/luci-static/resources/ui.js /www/luci-static/resources/ui.js
```

Then reload your browser / clear cache if needed

dhewg, 2023-02-09:
Any idea why all of these pops up now? Is that a browser or ib/auc issue?

jow-, 2023-02-09:
Because I accidentally committed a syntax error and people upgraded the affected luci-base package.

dhewg, 2023-02-09:
Oh okay, some of these issues made it sound like they were related to the 6g merge

jow-, 2023-02-10:
> Problem still exists with the latest snapshot from today.

Could be if it does not yet include the fix. The quickfix only works for people affected after a package upgrade, it will not solve bad images. Look out for an update to luci-base and install it available. In the meantime you can also try this shell command to remove the offending lines from ui.js:

```bash
sed -i -e '/> span >/d' /www/luci-static/resources/ui.js
```

Don't forget to clear your browser cache afterwards.

minax007, 2023-02-10:
>  it will not solve bad images. 

@jow- 

Everything else on the image is fine. 
It works in the background as expected (routing, adblock, ....). 
But I can't access Luci anymore.

minax007, 2023-02-13:
@jow- 

I can confirm that the problem was resolved with the latest snapshot version that was offered this morning. 

Thank you!