openwrt/luci #6232: snapshot version luci stuck at "Loading view..."

chinabb, 2023-02-09:
## Steps to reproduce:
Use the Chrome browser to access the router web interface, and the luci interface is stuck and a message "Loading view..." appears at any page I navigate and then hangs indefinitely.

## Actual behavior:

Chrome  F12 show errors:
GET http://192.168.1.1/cgi-bin/luci/ 403 (Forbidden)
luci.js?v=git-23.038.47117-2234d0c:161 SyntaxError: Unexpected identifier 'frameEl' (at luci.js?v=git-23.038.47117-2234d0c:155:3)
  in http://192.168.1.1/luci-static/resources/ui.js?v=git-23.038.47117-2234d0c:?
  at luci.js?v=git-23.038.47117-2234d0c:179:29
    at Object.raise (luci.js?v=git-23.038.47117-2234d0c:155:3)
    at luci.js?v=git-23.038.47117-2234d0c:179:29
luci.js?v=git-23.038.47117-2234d0c:161 SyntaxError: Unexpected identifier 'frameEl' (at luci.js?v=git-23.038.47117-2234d0c:155:3)
  in http://192.168.1.1/luci-static/resources/ui.js?v=git-23.038.47117-2234d0c:?
  at luci.js?v=git-23.038.47117-2234d0c:179:29
    at Object.raise (luci.js?v=git-23.038.47117-2234d0c:155:3)
    at luci.js?v=git-23.038.47117-2234d0c:179:29
luci.js?v=git-23.038.47117-2234d0c:155 Uncaught (in promise) SyntaxError: Unexpected identifier 'frameEl' (at luci.js?v=git-23.038.47117-2234d0c:155:3)
  in http://192.168.1.1/luci-static/resources/ui.js?v=git-23.038.47117-2234d0c:?
  at luci.js?v=git-23.038.47117-2234d0c:179:29
    at Object.raise (luci.js?v=git-23.038.47117-2234d0c:155:3)
    at luci.js?v=git-23.038.47117-2234d0c:179:29
raise @ luci.js?v=git-23.038.47117-2234d0c:155
(anonymous) @ luci.js?v=git-23.038.47117-2234d0c:179
Promise.then (async)
(anonymous) @ (index):58
luci.js?v=git-23.038.47117-2234d0c:155 Uncaught (in promise) SyntaxError: Unexpected identifier 'frameEl' (at luci.js?v=git-23.038.47117-2234d0c:155:3)
  in http://192.168.1.1/luci-static/resources/ui.js?v=git-23.038.47117-2234d0c:?
  at luci.js?v=git-23.038.47117-2234d0c:179:29
    at Object.raise (luci.js?v=git-23.038.47117-2234d0c:155:3)
    at luci.js?v=git-23.038.47117-2234d0c:179:29
raise @ luci.js?v=git-23.038.47117-2234d0c:155
(anonymous) @ luci.js?v=git-23.038.47117-2234d0c:179
Promise.then (async)
(anonymous) @ cbi.js?v=git-23.038.32982-f416c27:141
luci.js?v=git-23.038.47117-2234d0c:163 Uncaught (in promise) TypeError: classes.ui.addNotification is not a function
    at error (luci.js?v=git-23.038.47117-2234d0c:163:12)

## Expected behavior:

Use luci to manage openwrt router.

## Additional Information:
Using latest snapshot: luci master (git-23.009.82915-ec3aac4) / [OpenWrt SNAPSHOT r22004-047fb7c76d](https://downloads.openwrt.org/snapshots/targets/ipq40xx/generic/openwrt-ipq40xx-generic-asus_rt-ac42u-squashfs-sysupgrade.bin)
Kernel: [5.15.91-1-53b01a19f22d17f382ee562a60262f97]

OpenWrt version information from system `/etc/openwrt_release`

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r22004-047fb7c76d'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r22004-047fb7c76d'
DISTRIB_TAINTS=''

This is the log of installing luci after the first boot:

BusyBox v1.36.0 (2023-02-07 15:27:02 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r22004-047fb7c76d
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
[... 104 lines cut ...]
/luci-static/bootstrap
Configuring rpcd-mod-iwinfo.
Configuring luci-mod-status.
Configuring luci-app-firewall.
Configuring luci-proto-ppp.
Configuring luci-mod-network.
Configuring luci-mod-admin-full.
Configuring luci-proto-ipv6.
Configuring rpcd-mod-rpcsys.
Configuring attendedsysupgrade-common.
Configuring luci-app-attendedsysupgrade.
Configuring uhttpd.
4+0 records in
4+0 records out
Configuring rpcd-mod-rrdns.
Configuring uhttpd-mod-ubus.
4+0 records in
4+0 records out
Configuring luci-light.
Configuring luci.
root@OpenWrt:~# /etc/init.d/uhttpd enable
root@OpenWrt:~# /etc/init.d/uhttpd start
4+0 records in
4+0 records out
root@OpenWrt:~#

chinabb, 2023-02-09:
Supplementary information: 
Yesterday, using the same steps to install the previous day’s snapshot version was successful. I want to reinstall it today, but this error occurred. I have repeatedly flashed the firmware many times, and used the firstboot and reboot command to restore the factory and then install luci, but the error still occurs.

jow-, 2023-02-09:
Duplicate of #6230

closed 2023-02-09 by jow-