openwrt/luci #6233: luci-mod-status and luci-base: Luci Web UI stuck at Loading view after latest update

fab4x, 2023-02-09:
Yesterday I installed two Updates for my OpenWrt Router (luci-mod-status and luci-base) and now I can't access the Web Ui anymore. I already tried to reinstall luci but it didn't fix the problem. It just says "**Loading view...**". 
The Console log gives out the following error:

`luci.js?v=git-23.038.39289-18dde0c:155 Uncaught (in promise) SyntaxError: Unexpected identifier 'frameEl'
  in https://192.168.200.1/luci-static/resources/ui.js?v=git-23.038.39289-18dde0c:?
  at luci.js?v=git-23.038.39289-18dde0c:179:29
    at Object.raise (luci.js?v=git-23.038.39289-18dde0c:155:3)
    at luci.js?v=git-23.038.39289-18dde0c:179:29
luci.js?v=git-23.038.39289-18dde0c:163 Uncaught (in promise) TypeError: classes.ui.addNotification is not a function
    at error (luci.js?v=git-23.038.39289-18dde0c:163:12)`

Did anyone else experience the same?

Additional Infos about my Router:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0'
DISTRIB_REVISION='r19685-512e76967f'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0 r19685-512e76967f'
DISTRIB_TAINTS=''

minax007, 2023-02-09:
I have experienced the same with the latest snapshot build on an Asrock G10 (ipq806).

jow-, 2023-02-09:
Duplicate of https://github.com/openwrt/luci/issues/6230

closed 2023-02-09 by jow-