openwrt/luci #6241: Luci stuck at "Loading view..."

RewAU, 2023-02-11:
Since the most recent luci update, browser page is stuck at "Loading view..." at /cgi-bin/luci

Reviewing the browser page source suggests that luci.js now has two unhandled promise rejections in lines 872 and 898.

I was going to downgrade but thought this might be of interest ?

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.3'
DISTRIB_REVISION='r20028-43d71ad93e'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 22.03.3 r20028-43d71ad93e'
DISTRIB_TAINTS=''

RewAU, 2023-02-11:
I should probably also list versions of Luci and Lua;

opkg list-installed | grep luci

liblucihttp-lua - 2022-07-08-6e68a106-1
liblucihttp0 - 2022-07-08-6e68a106-1
luci - git-20.074.84698-ead5e81
luci-app-firewall - git-22.089.67563-7e3c1b4
luci-app-opkg - git-22.273.29004-9f6876b
luci-base - git-23.038.39289-18dde0c
luci-lib-base - git-20.232.39649-1f6dc29
luci-lib-ip - git-20.250.76529-62505bd
luci-lib-jsonc - git-22.097.61921-7513345
luci-lib-nixio - git-20.234.06894-c4a4e43
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-22.345.48602-4853e7b
luci-mod-status - git-23.038.33313-b256644
luci-mod-system - git-23.013.73113-588381e
luci-proto-ipv6 - git-21.148.48881-79947af
luci-proto-ppp - git-21.158.38888-88b9d84
luci-theme-bootstrap - git-22.288.45147-96ec0cd
rpcd-mod-luci - 20210614

opkg list-installed | grep lua

libiwinfo-lua - 2022-12-15-8d158096-1
liblua5.1.5 - 5.1.5-10
liblucihttp-lua - 2022-07-08-6e68a106-1
libubus-lua - 2022-06-01-2bebf93c-1
lua - 5.1.5-10

jow-, 2023-02-11:
Duplicate of https://github.com/openwrt/luci/issues/6230

closed 2023-02-11 by jow-