openwrt/luci #6255: luci-app-unbound: allow setting private-domain configuration

akostadinov, 2023-02-20:
Please allow setting one or more [private-domain](https://unbound.docs.nlnetlabs.nl/en/latest/manpages/unbound.conf.html#unbound-conf-private-domain) configuration directives to properly allows safer local network resolution.

## Steps to reproduce:

1. go to: Services → Recursive DNS → Advanced

## Actual behavior:
There is option to set `Filter Private Rebind` but no option to specify `private-domain` list.

## Expected behavior:
The should be an option to set `private-domain` list.

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='ipq806x/generic'
DISTRIB_ARCH='arm_cortex-a15_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''
```

DarkGhostHunter, 2024-07-02:
This is what breaks recursive name resolvers like `nip.io` and `sslip.io`.

Meanwhile, the option is to disable `Filter Private Rebind` and wait to be hacked :fearful:

systemcrash, 2024-07-06:
Changes here require changes in unbound startup modules to be effective. Search for the option there. Once you've added an option for `$UB_TXT_DOMAIN` then you can implement something in the GUI.

luisimasd, 2025-04-18:
Is this getting better on v24? or is it still present?