openwrt/luci #6257: Software tab missing

PerkelSimon, 2023-02-21:
Hi, 

Pretty easy to explain, in LuCi master, the `software tab` is missing since a while, is it possible to add it again?
thanks

systemcrash, 2023-02-22:
Did you bake a custom firmware?

PerkelSimon, 2023-02-22:
errror by myself, sorry ;-)

closed 2023-02-22 by PerkelSimon

mistepien, 2023-03-26:
So there is no "System->Software" if custom firmware is used?

mistepien, 2023-03-26:
OK, I see it now. I had to do it:
# opkg install luci-app-opkg

serdarcloud, 2023-04-04:
Hi, I have restored OpenWRT from one of the backups taken and I have also faced similar issue where the software tab is missing. 
I have run the below command and it give the below error. opkg update is working without any issue.
`root@OpenWrt:~# opkg install luci-app-opkg
Installing luci-app-opkg (git-22.273.29004-9f6876b) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/luci/luci-app-opkg_git-22.273.29004-9f6876b_all.ipk
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/luci/luci-app-opkg_git-22.273.29004-9f6876b_all.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download luci-app-opkg. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci-app-opkg.`
Kindly asaking for help if anyone knows the reason of the issue.
Thanks

hnyman, 2023-04-05:
you need to update your package lists again with "opkg update".
The package has been just rebuilt in the last buildbot build round...

https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/luci
It is there now:
https://downloads.openwrt.org/releases/22.03.2/packages/mips_24kc/luci/luci-app-opkg_git-23.093.42303-e16f620_all.ipk

luci-app-opkg_git-23.093.42303-e16f620_all.ipk | 9.8 KB | Tue Apr 4 21:47:25 2023

(You have likely just have had a bad timing, as you have tried it just when the buildbot is uploading new files, but there is not yet a new package index.)