openwrt/luci #6262: luci-mod-status: 160mhz channels are not properly displayed on the graph

abajk, 2023-02-24:
## Steps to reproduce:

1. go to: Status → Channel Analyzis -> radioX(5Ghz)
2. wait for the scan to finish

## Actual behavior:

Channels with 160 Mhz width are incorrectly parsed and are not in the graph.
Example below:
[image]

The error happens on Xiaomi AX3200. Maybe it is caused by HE mode (802.11ax). I do not have another supported device with WiFi AX to verify this theory.

## Expected behavior:

```
160 MHz network is correctly shown on the graph.
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.3'
DISTRIB_REVISION='r20028-43d71ad93e'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 22.03.3 r20028-43d71ad93e'
DISTRIB_TAINTS=''
```

abajk, 2023-02-24:
Output from iw scan command:
```
root@OpenWrt:~# iw wlan1 scan
BSS c0:3c:04:3a:f8:50(on wlan1) -- associated
        last seen: 437034.436s [boottime]
        TSF: 853731169240 usec (9d, 21:08:51)
        freq: 5520
        beacon interval: 100 TUs
        capability: ESS (0x1111)
        signal: -64.00 dBm
        last seen: 140 ms ago
        Information elements from Probe Response frame:
        SSID: Orange_Swiatlowod_F850
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
        HT capabilities:
                Capabilities: 0x1ef
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT RX MCS rate indexes supported: 0-31
                HT TX MCS rate indexes are undefined
        HT operation:
                 * primary channel: 104
                 * secondary channel offset: below
                 * STA channel width: any
        VHT capabilities:
                VHT Capabilities (0x0f8b69f5):
                        Max MPDU length: 7991
                        Supported Channel Width: 160 MHz
                        RX LDPC
                        short GI (80 MHz)
                        short GI (160/80+80 MHz)
                        TX STBC
                        SU Beamformer
                        MU Beamformer
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
        VHT operation:
                 * channel width: 1 (80 MHz)
                 * center freq segment 1: 106
[... 45 lines cut ...]
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                HE RX MCS and NSS set 160 MHz
                        1 streams: MCS 0-11
                        2 streams: MCS 0-11
                        3 streams: MCS 0-11
                        4 streams: MCS 0-11
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                HE TX MCS and NSS set 160 MHz
                        1 streams: MCS 0-11
                        2 streams: MCS 0-11
                        3 streams: MCS 0-11
                        4 streams: MCS 0-11
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                PPE Threshold 0x7b 0x1c 0xc7 0x71 0x1c 0xc7 0x71 0x1c 0xc7 0x71 0x1c 0xc7 0x71
...
```

jow-, 2023-02-25:
This should be fixed in master. The necessary iwinfo changes are not part of 22.03

closed 2023-02-25 by jow-

abajk, 2023-03-05:
Thanks for the hint. On my Xiaomi router WiFI is not working on the current master so I checked it on BT Home Hub 5A. The 160 Mhz channels are displayed as 80 MHz.

[image]

All networks with names starting with Orange Swiatlowod support 160MHz MIMO 4x4. My xiaomi router is connected to one of them (160 Mhz 2x2 link). So I'm sure there is some problem with parsing the 160 Mhz capability that is advertised by the AP.

~~~
root@OpenWrt:/# iw dev wlan0 scan
...
BSS c0:3c:04:3a:f8:50(on wlan0)
        last seen: 85.598s [boottime]
        TSF: 745140546564 usec (8d, 14:59:00)
        freq: 5520
        beacon interval: 100 TUs
        capability: ESS (0x1111)
        signal: -58.00 dBm
        last seen: 1850 ms ago
        Information elements from Probe Response frame:
        SSID: Orange_Swiatlowod_F850
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
        HT capabilities:
                Capabilities: 0x1ef
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT RX MCS rate indexes supported: 0-31
                HT TX MCS rate indexes are undefined
        HT operation:
                 * primary channel: 104
                 * secondary channel offset: below
                 * STA channel width: any
        VHT capabilities:
                VHT Capabilities (0x0f8b69f5):
                        Max MPDU length: 7991
                        Supported Channel Width: 160 MHz
                        RX LDPC
                        short GI (80 MHz)
                        short GI (160/80+80 MHz)
                        TX STBC
                        SU Beamformer
                        MU Beamformer
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
                VHT extended NSS: supported
        VHT operation:
                 * channel width: 1 (80 MHz)
                 * center freq segment 1: 106
                 * center freq segment 2: 114
                 * VHT basic MCS set: 0x0000
        HE capabilities:
                HE MAC Capabilities (0x010008120010):
                        Minimum Payload size of 128 bytes: 1
                        All Ack
                        Broadcast TWT
                        Maximum A-MPDU Length Exponent: 1
                        NDP Feedback Report
                HE PHY Capabilities: (0x4c2002c06f5b951800cc00):
                        HE160/HE80+80/5GHz
                        Punctured Preamble RX: 2
                        Doppler Rx
                        DCM Max Constellation: 3
                        DCM Max NSS Tx: 1
                        DCM Max Constellation Rx: 1
                        DCM Max NSS Rx: 1
                        Rx HE MU PPDU from Non-AP STA
                        Beamformee STS > 80Mhz: 6
                        Sounding Dimensions <= 80Mhz: 5
                        Sounding Dimensions > 80Mhz: 2
                        Ng = 16 MU Feedback
                        Codebook Size SU Feedback
                        Codebook Size MU Feedback
                        Triggered MU Beamforming Feedback
                        Triggered CQI Feedback
                        Partial Bandwidth DL MU-MIMO
                        80MHz in 160/80+80MHz HE PPDU
                        HE ER SU PPDU 1x HE-LTF 0.8us GI
                HE RX MCS and NSS set 80+80 MHz
                        1 streams: MCS 0-7
                        2 streams: not supported
                        3 streams: MCS 0-9
                        4 streams: MCS 0-7
                        5 streams: not supported
                        6 streams: MCS 0-11
                        7 streams: not supported
                        8 streams: MCS 0-9
                HE TX MCS and NSS set 80+80 MHz
                        1 streams: MCS 0-9
                        2 streams: MCS 0-7
                        3 streams: not supported
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: MCS 0-9
                        7 streams: MCS 0-7
                        8 streams: not supported
...
~~~

abajk, 2023-03-05:
@jow- From where does LUCI get the channel width data?

abajk, 2023-03-05:
Little endian (Mediatek MT7622/ARM Cortex A53):
~~~
                HE MAC Capabilities (0x010008120010):
~~~
Big endian (Lantiq VRX268/MIPS 34kc):
~~~
                HE MAC Capabilities (0x000112081000):
~~~
This looks like an endianess issue in the HE parsing code. Perhaps this is the primary cause.

abajk, 2023-03-05:
There are two problems:
* all tested devices do not correctly display 160 MHz channels on the chart, 
* atheros devices based on big endian architecture incorrectly display HE capabilitites (non-LUCI related error).

PR #7367 (2024-11-09): luci-mod-status: channel_analysis: detect 160 MHz capable AP
Implement a workaround to detect an 160MHz capable AP. It was introduced in the mac80211 in 2016 with 802.11ac Wave 2. APs capable of 160 MHz are detected by the shift of central frequencies. More detailed description in the link [1]. Every AP I have seen presents support for 160 MHz this way.

[1] https://github.com/torvalds/linux/commit/23665aaf9170ae6328cc4f68250c529a628af2ab
Fixes: #6262

[image]

- [x] This PR is not from my *main* or *master* branch :poop:, but a *separate* branch :white_check_mark:
- [x] Each commit has a vali [... cut]
commit aa79c3f3 (2024-11-09): luci-mod-status: channel_analysis: detect 160 MHz capable AP

Implement a workaround to detect an 160MHz capable AP. It was introduced
in the mac80211 in 2016 with 802.11ac Wave 2. APs capable of 160 MHz are
detected by the shift of central frequencies. More detailed description
in the link [1]. Every AP I have seen presents support for 160 MHz in this way.

[1] https://github.com/torvalds/linux/commit/23665aaf9170ae6328cc4f68250c529a628af2ab
Fixes: #6262