openwrt/luci #6265: [i18n] bad translation breaks luci

McGiverGim, 2023-02-28:
## Steps to reproduce:
1. Install `luci-i18n-https-dns-proxy-es` (the spanish translation of https-dns-proxy)
1. go to: Services→ Https dns proxy

## Actual behavior:
Luci crashes. 
[image]

## Expected behavior:
I want to load and see the https dns proxy page.

## Additional Information:
The problem is that one translation is bad, it was in the form `...(%smore...)` and by mistake it was translated by `(%more...)`. I have fixed it in weblate, but my suggestion is:
- Can we maybe catch the error, and ignore the bad translation, and use the original string? At least in this way luci work and is not break for one bad translation.
- Adding too an error at top of the page with the "bad translation" can help to find and fix the issue without breaking luci.

A second option:
- Simply modify the actual error page to show the bad string. This time I've found the bad string easily, but the next time maybe is not as easy.

hnyman, 2023-02-28:
I merged the weblate fix to master.  (I will backport to 22.03 later)

hnyman, 2023-02-28:
Fixed in master and 22.03

closed 2023-02-28 by hnyman

hnyman, 2023-02-28:
I am not sure if an easy/semi-easy solution for overcoming bad translation strings (bad print argument string formats etc.) can be easily spotted/fixed at runtime (e.g. by using the original English string if the exception originates from i18n.lua?), so I closed the issue after fixing the current bad string, although the intended discussion point is maybe wider.

If you @jow- think that this should be left open, feel free to reopen.