openwrt/luci #6280: Docker has only blank configuration

Bunnui, 2023-03-07:
Docker has only blank configuration

luci-lib-docker（Auto check）
luci-app-dockerman

[image]

## Additional Information:

Uptime：0h 6m 49s
Local Time：2023-03-08 03:55:38
Kernel Version：5.15.98
Model：VMware, Inc. VMware20,1
Architecture：Intel(R) Celeron(R) J4125 CPU @ 2.00GHz
Firmware Version：OpenWrt SNAPSHOT r22241-acb189179c

bauw2008, 2023-03-16:
Kernel Version: 5.15.98
The same problem, changed the author's source code PKG_VERSION:=v0.5.25, and modified it to display errors, and it can be used normally now.

jow-, 2023-04-29:
This seem to happen if the `dockerd` executable is not present / the dockerd package is not installed. The luci-app-dockerman module should probably depend on it.

reznyt, 2023-04-29:
I have dockerd installed and running, faced with the same issue.

Bunnui, 2023-04-29:
我试图完全删除docker，并离线安装luci应用程序dockerman的Release v0.5.25ipk，他给了我这个提示，这似乎是依赖内核版本的问题

```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nft-compat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-compat found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-extra
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-extra found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-ipt6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-ipt6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ip6tables
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ip6tables found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-nat6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-nat6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-conntrack
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-conntrack found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-nat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-nat found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-nat6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-nat6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-br-netfilter
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-br-netfilter found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-ipt-physdev
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-physdev found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-nf-ipvs
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-ipvs found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f) for kmod-veth
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-veth found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-dockerman:
 * 	kernel (= 5.15.105-1-e469f5589b4c7b368924a6e4f8f7407f)
 * opkg_install_cmd: Cannot install package luci-app-dockerman.
```

[image]

stokito, 2023-08-09:
I installed the dockerd but had a problem when the `/etc/config/dockerd` was already exists but empty and I had to replace it with opkg version `mv /etc/config/dockerd-opkg /etc/config/dockerd`. Then after reboot everything works fine.

kpoman, 2023-10-20:
> I installed the dockerd but had a problem when the `/etc/config/dockerd` was already exists but empty and I had to replace it with opkg version `mv /etc/config/dockerd-opkg /etc/config/dockerd`. Then after reboot everything works fine.

Worked as a charm

Yur0K, 2024-01-21:
Can anyone share the dockerd file (or contents of it)? I have the same issue with /etc/config/dockerdalready exists but empty. And dockerd-opkg is not there as well

hnyman, 2024-01-21:
> Can anyone share the dockerd file (or contents of it)?

It is in the package itself...
https://github.com/openwrt/packages/blob/master/utils/dockerd/files/etc/config/dockerd

closed 2024-10-08 by systemcrash