openwrt/luci #6281: Question: increase HTTP_MAX_CONTENT?

dibdot, 2023-03-07:
@jow- 
Currently HTTP_MAX_CONTENT is limited to 100kb ... this is very fast too little for the processing/editing of local lists in LuCI (e.g. in banIP). Would it be OK to increase this value let's say to 500 MB? Current browsers seem to support up to 2 GB ... ;-)
 
https://github.com/openwrt/luci/blob/e7afd0d327bb35c502ca41a3c5e3ea098898fbd7/modules/luci-base/ucode/http.uc#L17

trippleflux, 2023-03-07:
Probably also in [here](https://github.com/openwrt/luci/blob/master/libs/luci-lib-base/luasrc/http.lua#L14).

jow-, 2023-03-08:
*Browsers* will have no issues handling 2GB, your router will explode trying to handle that though...

dibdot, 2023-03-08:
The 2GB were not meant seriously ... maybe we could make the default configurable or the default would be derived from the available free space, e.g. 50% of the root.

jow-, 2023-03-09:
The HTTP post contents are held in memory, not stored on disk in most cases, so the max content would need to be inferred from available memory size.

Also note that the parsed HTTP post data contents in memory are larger than the raw data size, due to book keeping overhead.

closed 2023-12-30 by dibdot