openwrt/luci #6282: [BUG] lucihttp multipart-parser

PhilMeie, 2023-03-07:
I guess this is for jow only - or any other person that has knowledge of the internals of the lucihttp multipart-parser.

## Steps to reproduce:
1. cd /usr/lib/lua
2. lua
3. enter following lines in lua REPL (having multipart_parser trace enabled - stderr is marked with leading "-- "; I just added additional trace in case LH_MP_S_PART_DATA case (line 530))
  * `fprintf(stderr, "PART_DATA: off=%zu, offset=%zu, valuelen=%zu\n", off, p->offset, valuelen);` 
```
require "lucihttp"
what_txt={'BODY_BEGIN','PART_INIT','HEADER_NAME','HEADER_VALUE','PART_BEGIN','PART_DATA','PART_END','BODY_END','EOF'}

parser = lucihttp.multipart_parser("multipart/form-data; boundary=123456", function(what, buffer, length)
  print(what_txt[what+1], buffer, length)
	return true
  end, 2048)
parser:parse("--123456\r\n")
-- Buffer 3 (boundary 1) set data=(6)[123456]
-- Parsing buffer=(10)[--123456..]
-- Callback 0 (BODY_BEGIN) data=(6)[123456]
-- BODY_BEGIN      123456  6
-- State 0 (start of multipart body) -> 1 (start of boundary)
-- Callback 1 (PART_INIT) data=(0)[]
-- PART_INIT       nil     0
-- State 1 (start of boundary) -> 2 (start of header name)
parser:parse("Content-Disposition: form-data; name=\"binfile\"; filename=\"file.bin\"\r\nContent-Type: application/octet-stream")
-- Parsing buffer=(107)[Content-Disposition: form-data; name="binfile"; filename="file.bin"..Content-Type: application/octet-stream]
-- State 2 (start of header name) -> 3 (reading header name)
-- Buffer 0 (header name) append data=(19)[Content-Disposition]
-- State 3 (reading header name) -> 5 (start of header value)
-- State 5 (start of header value) -> 6 (reading header value)
-- Buffer 1 (header value) append data=(46)[form-data; name="binfile"; filename="file.bin"]
-- State 6 (reading header value) -> 7 (finding header value end)
-- State 7 (finding header value end) -> 2 (start of header name)
-- Callback 2 (HEADER_NAME) data=(19)[Content-Disposition]
-- HEADER_NAME     Content-Disposition     19
-- Callback 3 (HEADER_VALUE) data=(46)[form-data; name="binfile"; filename="file.bin"]
-- HEADER_VALUE    form-data; name="binfile"; filename="file.bin"  46
-- State 2 (start of header name) -> 3 (reading header name)
-- Buffer 0 (header name) append data=(12)[Content-Type]
-- State 3 (reading header name) -> 5 (start of header value)
-- State 5 (start of header value) -> 6 (reading header value)
-- Buffer 1 (header value) append data=(24)[application/octet-stream]
parser:parse("\r\n")
-- Parsing buffer=(2)[..]
-- State 6 (reading header value) -> 7 (finding header value end)
-- State 7 (finding header value end) -> 2 (start of header name)
parser:parse("\r\n")
-- Parsing buffer=(2)[..]
-- Callback 2 (HEADER_NAME) data=(12)[Content-Type]
-- HEADER_NAME     Content-Type    12
-- Callback 3 (HEADER_VALUE) data=(24)[application/octet-stream]
-- HEADER_VALUE    application/octet-stream        24
-- State 2 (start of header name) -> 3 (reading header name)
-- State 3 (reading header name) -> 4 (finding header name end)
-- State 4 (finding header name end) -> 8 (start of part data)
parser:parse("\x9e")
-- Parsing buffer=(3)[x9e]
-- Callback 4 (PART_BEGIN) data=(0)[]
-- PART_BEGIN      nil     0
-- State 8 (start of part data) -> 9 (reading part data)
-- PART_DATA: off=2, offset=0, valuelen=3
-- Buffer 2 (data) append data=(3)[x9e]
parser:parse("\r")  -- bad data (because ending with \r)
-- Parsing buffer=(1)[.]
-- PART_DATA: off=0, offset=0, valuelen=0
-- State 9 (reading part data) -> 10 (start of part boundary)
parser:parse("\r")
-- Parsing buffer=(1)[.]
-- Buffer 2 (data) append data=(2)[..]
-- State 10 (start of part boundary) -> 9 (reading part data)
parser:parse("\n")
-- Parsing buffer=(1)[.]
-- PART_DATA: off=0, offset=0, valuelen=1
-- Buffer 2 (data) append data=(1)[.]
parser:parse("-")
-- Parsing buffer=(1)[-]
-- PART_DATA: off=0, offset=0, valuelen=1
-- Buffer 2 (data) append data=(1)[-]
parser:parse("-")
-- Parsing buffer=(1)[-]
-- PART_DATA: off=0, offset=0, valuelen=1
-- Buffer 2 (data) append data=(1)[-]
parser:parse("12345")
-- Parsing buffer=(5)[12345]
-- PART_DATA: off=4, offset=0, valuelen=5
-- Buffer 2 (data) append data=(5)[12345]
parser:parse("6")
-- Parsing buffer=(1)[7]
-- PART_DATA: off=0, offset=0, valuelen=1
-- Buffer 2 (data) append data=(1)[6]
parser:parse("-")
-- Parsing buffer=(1)[-]
-- PART_DATA: off=0, offset=0, valuelen=1
-- Buffer 2 (data) append data=(1)[-]
parser:parse("-")
-- Parsing buffer=(1)[-]
-- PART_DATA: off=0, offset=0, valuelen=1
-- Buffer 2 (data) append data=(1)[-]
parser:parse("\r")
-- Parsing buffer=(1)[.]
-- PART_DATA: off=0, offset=0, valuelen=0
-- State 9 (reading part data) -> 10 (start of part boundary)
parser:parse("\n")
-- Parsing buffer=(1)[.]
-- State 10 (start of part boundary) -> 11 (reading part boundary)
```

## Actual behavior:
* If multipart binary file ends with "\r" (aka 0x0d) parser changes to wrong state and does not recover anymore

## Expected behavior:
* any byte should be allowed in multipart binary files

## Additional Information:
1. If the binary data does not end with "\r", then everything works fine
2. Example (with same lucihttp.multipart_parser as above)
```
parser = lucihttp.multipart_parser("multipart/form-data; boundary=123456", function(what, buffer, length)
  print(what_txt[what+1], buffer, length)
	return true
  end, 2048)
parser:parse("-")
-- Callback 0 (BODY_BEGIN) data=(6)[123456]
-- BODY_BEGIN      123456  6
-- State 0 (start of multipart body) -> 1 (start of boundary)
parser:parse("-123456\r")
-- Parsing buffer=(8)[-123456.]
parser:parse("\n")
-- Parsing buffer=(1)[.]
-- Callback 1 (PART_INIT) data=(0)[]
-- PART_INIT       nil     0
-- State 1 (start of boundary) -> 2 (start of header name)
parser:parse("Content-Disposition: form-data; name=\"binfile\"; filename=\"file.bin\"\r\nContent-Type: application/octet-stream")
-- Parsing buffer=(107)[Content-Disposition: form-data; name="binfile"; filename="file.bin"..Content-Type: application/octet-stream]
-- State 2 (start of header name) -> 3 (reading header name)
-- Buffer 0 (header name) append data=(19)[Content-Disposition]
-- State 3 (reading header name) -> 5 (start of header value)
-- State 5 (start of header value) -> 6 (reading header value)
-- Buffer 1 (header value) append data=(46)[form-data; name="binfile"; filename="file.bin"]
-- State 6 (reading header value) -> 7 (finding header value end)
-- State 7 (finding header value end) -> 2 (start of header name)
-- Callback 2 (HEADER_NAME) data=(19)[Content-Disposition]
-- HEADER_NAME     Content-Disposition     19
-- Callback 3 (HEADER_VALUE) data=(46)[form-data; name="binfile"; filename="file.bin"]
-- HEADER_VALUE    form-data; name="binfile"; filename="file.bin"  46
-- State 2 (start of header name) -> 3 (reading header name)
-- Buffer 0 (header name) append data=(12)[Content-Type]
-- State 3 (reading header name) -> 5 (start of header value)
-- State 5 (start of header value) -> 6 (reading header value)
-- Buffer 1 (header value) append data=(24)[application/octet-stream]
parser:parse("\r\n")
-- Parsing buffer=(2)[..]
-- State 6 (reading header value) -> 7 (finding header value end)
-- State 7 (finding header value end) -> 2 (start of header name)
parser:parse("\r\n")
-- Parsing buffer=(2)[..]
-- Callback 2 (HEADER_NAME) data=(12)[Content-Type]
-- HEADER_NAME     Content-Type    12
-- Callback 3 (HEADER_VALUE) data=(24)[application/octet-stream]
-- HEADER_VALUE    application/octet-stream        24
-- State 2 (start of header name) -> 3 (reading header name)
-- State 3 (reading header name) -> 4 (finding header name end)
-- State 4 (finding header name end) -> 8 (start of part data)
parser:parse("\x9e")
-- Parsing buffer=(3)[x9e]
-- Callback 4 (PART_BEGIN) data=(0)[]
-- PART_BEGIN      nil     0
-- State 8 (start of part data) -> 9 (reading part data)
-- PART_DATA: off=2, offset=0, valuelen=3Buffer 2 (data) append data=(3)[x9e]
parser:parse("?")  -- good data
-- Parsing buffer=(1)[?]
-- PART_DATA: off=0, offset=0, valuelen=1Buffer 2 (data) append data=(1)[?]
parser:parse("\r")
-- Parsing buffer=(1)[.]
-- PART_DATA: off=0, offset=0, valuelen=0State 9 (reading part data) -> 10 (start of part boundary)
parser:parse("\n")
-- Parsing buffer=(1)[.]
-- State 10 (start of part boundary) -> 11 (reading part boundary)
parser:parse("--12345")
-- Parsing buffer=(7)[--12345]
parser:parse("6")
-- Parsing buffer=(1)[6]
-- Callback 5 (PART_DATA) data=(4)[x9e?]
-- PART_DATA       x9e?    4
-- Callback 6 (PART_END) data=(0)[]
-- PART_END        nil     0
-- State 11 (reading part boundary) -> 12 (finding part boundary end)
parser:parse("-")
-- State 12 (finding part boundary end) -> 14 (end of final part)
parser:parse("-")
-- Parsing buffer=(1)[-]
-- Callback 7 (BODY_END) data=(6)[123456]
-- BODY_END        123456  6
-- State 14 (end of final part) -> 15 (end of multipart body)
parser:parse("\r")
-- Parsing buffer=(1)[.]
parser:parse("\n")
-- Callback 8 (EOF) data=(0)[]
-- EOF     nil     0
```

PhilMeie, 2023-03-08:
I propose the following patch for lucihttp multipart-parser.
[001-fix-multipart-parser.patch](https://github.com/openwrt/luci/files/10920021/001-fix-multipart-parser.patch)

jow-, 2023-03-15:
@PhilMeie - thank you for your detailed analysis and the proposed path, I pushed a refactored version of your change and added a testcase for this particular bug.

closed 2023-03-15 by jow-

PhilMeie, 2023-03-15:
> @PhilMeie - thank you for your detailed analysis and the proposed path, I pushed a refactored version of your change and added a testcase for this particular bug.

@jow- - Great, thanks for the feedback and refactoring.

commit 0f6e1663 (2023-03-15): lucihttp: update to latest Git HEAD

9b5b683 multipart-parser: properly handle parts with trailing CR

Fixes: #6282
commit d927ed5a (2023-04-03): lucihttp: update to latest Git HEAD

9b5b683 multipart-parser: properly handle parts with trailing CR

Fixes: #6282

(cherry picked from commit 0f6e1663ca0370ab3f8ca3927a9d0a37b015d0f4)