openwrt/luci #6284: luci-app-attendedsysupgrade: incomplete package list shown #6049

edrikk, 2023-03-08:
## Steps to reproduce:

1. Attended Sysupgrade should be put into "advanced" mode, so as to allow package selection.
2. Click the "Search for firmware upgrade" button
3. On the "New firmware upgrade available" page you can see description of packages.  Click "Request firmware image"
4. On the "Successfully created firmware image" page click *Cancel*.  You will be returned to the main "Attended Sysupgrade" page
5. Without refreshing page or browsing away, click the "Search for firmware upgrade" button

## Actual behavior:
Upon re-entering the "New firmware upgrade available" lightbox / page, instead of seeing the description of the packages, the user is shown numbers in the list.

This behaviour is device (AX3600, AX6S, R6700) and browser (Safari, Chrome, Firefox) independent.

## Expected behavior:
Re-entering the "New firmware upgrade available" after cancelling the first time should again show the package descriptions.

Note:  An F5 refresh, or browsing away and returning and then clicking on the "Search for firmware upgrade" will again correctly show the descriptions.  However, repeating the above steps will result in the same (incorrect) behaviour.

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r22188-7662700613'
DISTRIB_TARGET='ipq807x/generic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r22188-7662700613'
DISTRIB_TAINTS=''
```

McGiverGim, 2023-03-10:
I noticed the same problem when the build fails for some reason (wrong package selection or similar). So no only happens when cancelling.

jow-, 2023-06-07:
@aparcar

edrikk, 2023-10-12:
To ensure this doesn't go stale, noting that this issue is still there as of October 11, 2023 snapshot.

aparcar, 2023-10-12:
Yea I'm aware of this issue but don't know what the issue here, maybe @jow- got a clue why copying a dict (packages) twice causes this issue.

dannil, 2024-02-28:
I have successfully identified what the problem is and implemented a solution, there'll be a PR in the upcoming days.

dannil, 2024-03-02:
Feel free to review #6952 which fixes this.

closed 2024-03-19 by systemcrash