openwrt/luci #6291: Missing some IPv4 info in the Associated Stations list

RafcioRS, 2023-03-14:
I'm running OpenWRT (OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-22.361.69894-438c598) on Tp-Link Archer A7 V5 hardware in "dumb" AP mode. I've found a couple of bugs and here is a description of the second of them.

Host info for some (well, most, but not all) entries is missing IPv4 address. All of the hosts (obviously) have IPv4 address, but not all have IPv6. IPv6 is shown for all hosts that have it, but those that don't have IPv6 are shown with question mark as IPv4 is missing. It seems like some "reverse ARP" issue. The MAC address is correctly present though for all entries. The reverse DNS (host name) is working for those (few) hosts that have IPv4 address.
 
I'm not sure what else I can add here. If there is any helpful logging that needs to be enabled and the logs attached, I'm willing to do that.

jow-, 2023-03-15:
The router is simply not aware of these hosts IP address as it is handling their traffic on layer 2. LuCI itself does not perform active ARP probing or host discovery, it simply passively reports what is known to the kernel.

This is a well known limitations of dumb APs. Run the DHCP server on your OpenWrt system or implement some form of periodic host discovery to make these entries appear in LuCI.

closed 2023-03-15 by jow-

RafcioRS, 2023-03-16:
OK, so there is no module performing ARPing in the background. How come though all IPV6 addresses are associated with MAC addresses, and only a few (well less than half) of IPv4 addresses are known? Shouldn't IPv6 addresses be impacted in the same way?

RafcioRS, 2023-03-22:
All right. I still don't know why IPv4 addresses are missing, but IPv6 ones are not. I've developed a workaround/solution and I'm sharing it here for others to benefit.
I've installed arp-scan package, but running it by itself didn't do anything, even though it was getting all active MAC/IP pairs correctly. I had to ping each IP in order for kernel to be aware of it. Again, I don't know why is that.
Here is my startup script (/etc/rc.local) addition:
/usr/bin/arp-scan -I br-lan -l -x -g | /usr/bin/awk "{system(\"/bin/ping -c 5 \" \$1)}" 1>dev/nul &
And I also run the command through cron every hour (under Scheduled Tasks):
0 * * * * /usr/bin/arp-scan -I br-lan -l -x -g | /usr/bin/awk "{system(\"/bin/ping -c 5 \" \$1)}" 1>dev/nul

RafcioRS, 2023-05-05:
For others to benefit, here is a correction to the crontab entry.
0 * * * * sh -c "arp-scan -I br-lan -l -x -g | awk '{system(\"ping -c 5 \" \$1)}'" >/tmp/arp-scan-awk-cron.log 2>&1
Also, startup entry is not needed. It's too early in the process and cron entries are run anyway, regardless of the time.