openwrt/luci #6298: luci-app-acme: "Apply to nginx" does not work.

Curious-r, 2023-03-19:
***

## Steps to reproduce:

1. Install luci-app-acme by offical feed
2. Modify the certificate configuration and check "Apply to nginx"
3. Save and apply

## Actual behavior:

1. Certs was successfully issued in `/etc/acme`, but it was not applied to nginx.

## Expected behavior:

1. Certs apply to nginx rightly

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r0-e467856'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r0-e467856'
DISTRIB_TAINTS='no-all'
```
The firmware is with luci-ssl-nginx

Curious-r, 2023-03-19:
Well,I seem to remember that I solved this problem two years ago. 
We need to manually create the UCI configuration corresponding to the domain (or nginx native configuration) before ACME can help us update the certificate configuration.

Curious-r, 2023-03-19:
No.. This didn't work out, as if "apply" wouldn't trigger at all.

jow-, 2023-03-20:
LuCI merely modifies the configuration for the underlying acme package, so the issue likely lies there. @tohojo ?