openwrt/luci #6300: luci-app-dockerman; overview page error

Gost6, 2023-03-20:
Hello,

I have noticed an error on the luci-app-dockerman/overview page, this happen with latest docker bump to 23.0.1.

[image]

DISTRIB_ID="OpenWrt'
DISTRIB_RELEASE='master'
DISTRIB_REVISION='r22325-00b40ef00d'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
```

vbzc, 2023-04-06:
same issues

closed 2023-04-28 by jow-

kimboslice99, 2024-03-26:
Did this fix not make it into the 23.05.3 release? Still an issue in 23.05.3

realJustinLee, 2024-03-26:
Still existing in 2023.05.3

systemcrash, 2024-03-26:
Paste a copy of the trace-back text here. Look very closely to see whether they match. This fix is in 23.

kimboslice99, 2024-03-26:
It is slightly different, here's the full trace... should I open a new issue for this?

```
/usr/lib/lua/luci/ucodebridge.lua:23: ...sr/lib/lua/luci/model/cbi/dockerman/newcontainer.lua:311: attempt to concatenate a nil value

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:794:4)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:795:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1003:48)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^
```

systemcrash, 2024-03-26:
> It is slightly different, here's the full trace... should I open a new issue for this?

This not being the same issue, yes.

Include your dockerman config, and what you tried to do.

commit 39d387f3 (2023-04-28): luci-app-dockerman: gracefully handle empty registry mirror setting

Fixes: #6300
commit 08fd2d54 (2023-04-28): luci-app-dockerman: gracefully handle empty registry mirror setting

Fixes: #6300

(cherry picked from commit 39d387f34c083afab9589952c6bcc9a0028cc9da)