openwrt/luci #6337: luci-app-babeld: Bug/Feature request: Make interface readable

marek22k, 2023-04-11:
## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Status → Babeld tab
2. look at the status

## Actual behavior:
A very ugly site. I think the whole thing is supposed to be a table, but it is not.

## Expected behavior:
Something clear - for example, a table.

## Additional Information:

I use the Luci Design OpenWrt2020. I have attached pictures.

OpenWrt version information from system `/etc/openwrt_release`

**Example: Replace the following lines and remove this**
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.3'
DISTRIB_REVISION='r20028-43d71ad93e'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.3 r20028-43d71ad93e'
DISTRIB_TAINTS=''
```

[image]

marek22k, 2023-04-11:
I think https://github.com/openwrt/luci/blob/master/applications/luci-app-babeld/root/www/luci-static/resources/babeld.js#L38 is supposed to be a table. However, div elements are used for this. Maybe that is the reason for the appearance?

marek22k, 2023-04-11:
For the Bootstrap and OpenWrt themes, a table actually appears.

JonnyTischbein, 2023-04-13:
At first it was build using the table tag (see https://github.com/openwrt/luci/pull/3927#issuecomment-615781467 ) but changed to the "common div structure" because table appears to be deprecated.

The theme does not seem to support this: https://github.com/openwrt/luci/blob/openwrt-22.03/themes/luci-theme-openwrt-2020/htdocs/luci-static/openwrt2020/cascade.css
Maybe @jow- knows something about it ?

marek22k, 2023-04-13:
Why is table obsolete? As far as I know, the element still exists in HTML5.

commit 70fb460c (2023-04-28): luci-app-babeld: use table markup for overview tables

Fixes: #6337
commit edcd682a (2023-04-28): luci-app-babeld: use table markup for overview tables

Fixes: #6337

(cherry picked from commit 70fb460c00d6c641a39f0cf97eb7eb846bee51e2)
closed 2023-04-28 by jow-