openwrt/luci #6342: luci-app-babeld: Hello-Reach is represented as a decimal number.

marek22k, 2023-04-13:
## Steps to reproduce:

1. go to: Status → Babeld tab
2. look at the status

## Actual behavior:
Hello-Reach is represented as a hexadecimal number or with checkboxes.

## Additional Information:

I also asked on the Babel Mailling list if this representation makes sense: https://alioth-lists.debian.net/pipermail/babel-users/2023-April/004081.html

OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.3'
DISTRIB_REVISION='r20028-43d71ad93e'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.3 r20028-43d71ad93e'
DISTRIB_TAINTS=''
```

jow-, 2023-04-28:
Can't find any formatting logic in the LuCI app, it outputs the data as received from ubus:

https://github.com/openwrt/luci/blob/master/applications/luci-app-babeld/root/www/luci-static/resources/babeld.js#L145

The ubus emitter does no decimal formatting either:

https://github.com/openwrt/routing/blob/master/babeld/src/ubus.c#L355

Can you append a screenshot? Also the outputs of the following commands:
 - `ubus call babeld get_info` 
 - `ubus call babeld get_xroutes` 
 - `ubus call babeld get_routes` 
 - `ubus call babeld get_neighbours`

PolynomialDivision, 2023-04-28:
```
root@rhnk-core:~# ubus call babeld get_info
{
	"babeld-version": "babeld-1.12.2-ubus-mod",
	"my-id": "ea:9f:80:ff:fe:5d:30:82",
	"host": "rhnk-core"
}
```
```
root@rhnk-core:~# ubus call babeld get_xroutes
{
	"IPv4": {
		
	},
	"IPv6": {
		"2001:bf7:820:1000::/56": {
			"src-prefix": "::/0",
			"metric": 0
		}
	}
}
```
```
root@rhnk-core:~# ubus call babeld get_routes | head -n 40
{
	"IPv4": {
		
	},
	"IPv6": {
		"2001:bf7:860:d00::/56": {
			"src-prefix": "::/0",
			"route_metric": 1393,
			"route_smoothed_metric": 1420,
			"refmetric": 608,
			"id": "46:4e:6d:ff:fe:8a:de:e5",
			"seqno": 35970,
			"channels": " chan (255)",
			"age": 10,
			"via": "fe80::ea9f:80ff:fe5d:39d1",
			"installed": false,
			"feasible": false
		},
		"2001:bf7:860:d00::/56": {
			"src-prefix": "::/0",
			"route_metric": 2656,
			"route_smoothed_metric": 2660,
			"refmetric": 1888,
			"id": "46:4e:6d:ff:fe:8a:de:e5",
			"seqno": 35970,
			"channels": " chan (255)",
			"age": 6,
			"via": "fe80::ea9f:80ff:fe5d:30c5",
			"installed": false,
			"feasible": false
		},
		....
```
```
root@rhnk-core:~# ubus call babeld get_neighbours
{
	"IPv4": {
		
	},
	"IPv6": {
		"fe80::ea9f:80ff:fe5d:30c5": {
			"dev": "switch0.24",
			"hello-reach": 65535,
			"uhello-reach": 0,
			"rxcost": 256,
			"txcost": 256,
			"rtt": "0.000",
			"channel": -2,
			"if_up": true
		},
		"fe80::6666:b3ff:fedf:554": {
			"dev": "switch0.32",
			"hello-reach": 65535,
			"uhello-reach": 0,
			"rxcost": 256,
			"txcost": 256,
			"rtt": "0.000",
			"channel": -2,
			"if_up": true
		},
		"fe80::2e3a:fdff:fe25:152b": {
			"dev": "switch0.14",
			"hello-reach": 65535,
			"uhello-reach": 0,
			"rxcost": 256,
			"txcost": 256,
			"rtt": "0.000",
			"channel": -2,
			"if_up": true
		},
		"fe80::ea9f:80ff:fe5d:31a1": {
			"dev": "switch0.11",
			"hello-reach": 65535,
			"uhello-reach": 0,
			"rxcost": 256,
			"txcost": 256,
			"rtt": "0.000",
			"channel": -2,
			"if_up": true
		},
		"fe80::6222:32ff:fe59:51bf": {
			"dev": "switch0.23",
			"hello-reach": 65535,
			"uhello-reach": 0,
			"rxcost": 256,
			"txcost": 256,
			"rtt": "0.000",
			"channel": -2,
			"if_up": true
		},
		"fe80::ea9f:80ff:fe5d:39d1": {
			"dev": "switch0.15",
			"hello-reach": 65407,
			"uhello-reach": 0,
			"rxcost": 257,
			"txcost": 256,
			"rtt": "0.000",
			"channel": -2,
			"if_up": true
		}
	}
}
```

I need to adjust the ubus bindings to fix duplicate entries:
https://github.com/openwrt/routing/issues/966

marek22k, 2023-04-28:
[image]

PolynomialDivision, 2023-04-28:
Let's change reach and ureach to an array of bools in the ubus bindings.

closed 2024-09-08 by PolynomialDivision