openwrt/luci #6349: [Question] I have something to say about the difference between the official version of OpenWRT and the Chinese version

8MiYile, 2023-04-17:
This topic comes from a project called [Lean's OpenWRT](https://github.com/coolsnowwolf/lede) that I found on the GitHub Chinese board
Their ecology in terms of applications is really much better than the official version of OpenWRT, and there are many more luci-apps than the official version.
This is their [LuCI Repo](https://github.com/coolsnowwolf/luci)
The question I've always been curious about is why your officials don't look at this branch of the Repo
You guys can recall LEDE and merge it into your mainline repo, why can't you also merge Lean's OpenWRT into your mainline repo and test it?
My hope is that, since they are all OpenWRT, there will be no distinction between the Lean version and the LEDE version.
Because users who use it will have headaches saying why each version is very different, why the number of software in the official version is not as good as the Lean version.
More simply put, it doesn't matter who released it or who compiled it, as long as it doesn't affect subsequent software and firmware updates, it's enough

hnyman, 2023-04-17:
There is a big difference: LEDE was pretty much the whole active OpenWrt team, which first split off as LEDE and then later merged back and took over the previous OpenWrt repo.

closed 2023-04-17 by hnyman

8MiYile, 2023-04-17:
So you guys just don't try to fix it? Why just shut it down?

8MiYile, 2023-04-17:
@hnyman hello?

hnyman, 2023-04-17:
No new apps are pulled in from third-party repositories. If the author wants to submit them here via PRs, that is fine.

8MiYile, 2023-04-17:
@hnyman If you still allow the existence of the language lua, otherwise it is very difficult