openwrt/luci #6360: luci: Missleading error message when rpcd fails

emhamm, 2023-04-27:
## Steps to reproduce:

1. Have ~5k active connections
2. go to: Status → Realtime Graphs → Connections
3. wait a few seconds
4. reload page because it will not respond anymore
5. see error message `/usr/lib/lua/luci/dispatcher.lua:431: /etc/config/luci seems to be corrupt, unable to find section 'main'`

## Actual behavior:
* Error message `/usr/lib/lua/luci/dispatcher.lua:431: /etc/config/luci seems to be corrupt, unable to find section 'main'`

## Expected behavior:
* Maybe `rpcd does not respond`? 

## Additional Information:
At least 22.03.[0,1,2,3] is also effected, 21.03 probably too. See https://github.com/openwrt/luci/issues/4758 for reference.

OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.4'
DISTRIB_REVISION='r20123-38ccc47687'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 22.03.4 r20123-38ccc47687'
```

xenago, 2023-06-12:
Can confirm this, I'm running 22.03.5 and it is still a problem

dongc123, 2023-07-28:
目标平台 | x86/64
固件版本 | OpenWrt 23.05.0-rc2
内核版本 | 5.15.118
There is also this bug
[image]
[image]
[image]
[image]