openwrt/luci #6377: luci-app-travelmate: generates wrong QR codes for WPA3 networks

patrakov, 2023-05-06:
## Steps to reproduce:

1. go to: Network → Wireless
2. create a WPA3 SAE or WPA3 SAE / WPA2 PSK mixed network
3. go to: Services → Travelmate (luci-app-travelmate version git-22.232.72147-7971fe0)
4. Press the "AP QR-codes" button.
5. Generate a QR code for the newly created network
6. Scan it with an Android phone

## Actual behavior:

"Check connection and try again" red message in the QR scanner (at least on a Poco X4 Pro 5G phone)

## Expected behavior:

Phone should connect to the WiFi

## Additional Information:

luci-app-travelmate inserts an invalid "T:" tag in the generated QR codes. It is set to "sae" or "sae-mixed", exactly as written in /etc/config/wireless, while the correct value is "WPA" according to the WPA3 specification (but this doesn't actually work) or "SAE" (when attempting to reshare the WiFi credentials from the phone).

In other words, please map the "sae" or "sae-mixed" encryption types to "SAE" (in capital letters).

closed 2023-05-08 by dibdot

patrakov, 2023-05-08:
Not fixed. Even though "WPA" is the correct value per the spec, only "SAE" works with Poco X4 Pro 5G. The error is "Check connection and try again".

More precisely, WPA2 + WPA3 mixed mode works, but pure WPA3 doesn't.

dibdot, 2023-05-08:
Then your mobile phone seems to be ignoring official standards. I've tested this change successfully with my mobile phone (Android 12) and a 0815 china tablet (Android 10).

patrakov, 2023-05-08:
It's not only my phone. The same issue is present in the Google-provided AOSP sources. Here is a function that returns the string for the "T:" tag when one shares the network from the phone:

https://android.googlesource.com/platform/packages/apps/Settings/+/refs/heads/master/src/com/android/settings/wifi/dpp/WifiDppUtils.java#164

Copy-pasting:

```java
    static String getSecurityString(WifiConfiguration config) {
        if (config.allowedKeyManagement.get(KeyMgmt.SAE)) {
            return WifiQrCode.SECURITY_SAE;
        }
        if (config.allowedKeyManagement.get(KeyMgmt.OWE)) {
            return WifiQrCode.SECURITY_NO_PASSWORD;
        }
        if (config.allowedKeyManagement.get(KeyMgmt.WPA_PSK) ||
                config.allowedKeyManagement.get(KeyMgmt.WPA2_PSK)) {
            return WifiQrCode.SECURITY_WPA_PSK;
        }
        return (config.wepKeys[0] == null) ?
                WifiQrCode.SECURITY_NO_PASSWORD : WifiQrCode.SECURITY_WEP;
    }
```

Where WifiQrCode.SECURITY_SAE is defined as "SAE" here: https://android.googlesource.com/platform/packages/apps/Settings/+/refs/heads/master/src/com/android/settings/wifi/dpp/WifiQrCode.java#70

My point is that the standard is worthless if it is not followed by Google itself.