openwrt/luci #6382: luci-mod-network static leases not bound to specific instance

professor-jonny, 2023-05-08:
When configuting a static lease from luci it does not bind it to a specific instance.

Manual editing of the config files is required to acomplish this.

professor-jonny, 2023-08-09:
As I believe multiple dnsmasq instances are not supported well in luci and there is various other issues.

dave14305, 2023-09-13:
Assigning static leases to instances was added in https://github.com/openwrt/luci/commit/f73abffa2aa64a91e6c16ddae7966a5c35e73fe1 in snapshots, but the implementation seems broken because the `instance` value should be the name of the `config dnsmasq` section and not an index number.

Pinging @systemcrash or @jow- to validate my findings. Discussed [in the forum](https://forum.openwrt.org/t/dnsmasq-might-be-broken/171255/23?u=dave14305) recently.

systemcrash, 2023-09-13:
The GUI additions in f73abffa2aa64a91e6c16ddae7966a5c35e73fe1 ask luci to poll for the available sections, so those are available to choose from if they exist. I think other sections only appear in manually edited configs anyhow, since the GUI isn't yet set up for multiple dnsmasq instances.

professor-jonny, 2023-09-13:
@dave14305 they seem bound to the correct instance by config in my setup as, I can create static leases for the same mac addresses in different lan segments without issue it is just that luci reports them as the name of the first instance in the DHCP config file.