openwrt/luci #6391: <luci-app-hd-idle>:HDD was actived when entered the luci-app-hd-idle from luci web every time.

chunfengyao, 2023-05-16:
## Steps to reproduce:
- install `luci-app-hd-idle`
- plug in the HDD
- wait for hd-idle to sleep the HDD
- open luci web, navigate to `/cgi-bin/luci/admin/services/hd_idle`
- then the HDD will been activated

## Actual behavior:
- the HDD was activated

## Expected behavior:
- the HDD was still in sleeping

## Additional Information:
```ini
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.5'
DISTRIB_REVISION='r20134-5f15225c1e'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 22.03.5 r20134-5f15225c1e'
DISTRIB_TAINTS='no-all busybox'
```

According to the info that `luci-app-hd-idle`'s web page shows me, possibly the cause of this problem is that `luci-app-hd-idle` try to read the vendor and the model of the HDD. by [this pr](https://github.com/openwrt/luci/pull/5444).

psiegl, 2023-05-16:
I doubt that the PR I posted has an impact on the idling of the hdds. But let's try.
If you check the code, in line 27 of hd_idle.js, the sole addition is the following command:

lsblk -n -J -do NAME,TRAN,ROTA,RM,VENDOR,MODEL

In case your hdds are idling, can you run this cmd and check if they come online again?
If yes, it is the PR, otherwise it can't, because anything else is just a bit of UI prettifying.

chunfengyao, 2023-05-17:
I test with an old hdd.
hdd comes online after excute `lsblk -n -J -do NAME,TRAN,ROTA,RM,VENDOR,MODEL`.
I prefer to obtain device information through the use of `smartctl`.

here's the shell script and log file. we can see that the `Power mode` has changed to 'ACTIVE or IDLE' after executing `lsblk -n -J -do NAME,TRAN,ROTA,RM,VENDOR,MODEL` at `Wed May 17 00:18:36 UTC 2023`.

### script
```shell
#!/bin/sh
set -x
date -u ;ps -ef | grep hd-|grep -v grep
sleep 6s
date -u ;smartctl -n standby  -i /dev/sdb
date -u ;lsblk -n -J -do NAME,TRAN,ROTA,RM,VENDOR,MODEL /dev/sdb
date -u ;smartctl -n standby  -i /dev/sdb
sleep 6s
date -u ;smartctl -n standby  -i /dev/sdb
smartctl --json -n standby  -i /dev/sdb
date -u ;smartctl -n standby  -i /dev/sdb

```

### logs

```shell
Microsoft Windows [版本 10.0.22621.1776]
(c) Microsoft Corporation。保留所有权利。

C:\Users\14325>openwrt

BusyBox v1.35.0 (2023-04-27 20:28:15 UTC) built-in shell (ash)

 _   _
| | | | ___  _ __ ___   ___
| |_| |/ _ \| '_ ` _ \ / _ \
|  _  | (_) | | | | | |  __/
|_| |_|\___/|_| |_| |_|\___|
  ___                                _
 / _ \ _ __   ___ _ ____      ___ __| |_
| | | | '_ \ / _ \ '_ \ \ /\ / / '__| __|
| |_| | |_) |  __/ | | \ V  V /| |  | |_
 \___/| .__/ \___|_| |_|\_/\_/ |_|   \__|
      |_|W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt Built By chunfeng.yao
 -----------------------------------------------------
Login at Wed May 17 08:18:07 CST 2023 from 10.0.0.15 6202
root@OpenWrt:~# /tmp/powermodeofHDD.sh
+ date -u
Wed May 17 00:18:26 UTC 2023
+ grep hd-
+ ps -ef
+ grep -v grep
root     16279 11460  0 08:13 pts/8    00:00:00 /usr/bin/hd-idle -d -i 0 -a sda -i 300 -a sdb -i 5 -a sdc -i 300 -a sdd -i 300 -a sde -i 300
+ sleep 6s
+ date -u
Wed May 17 00:18:32 UTC 2023
+ smartctl -n standby -i /dev/sdb
smartctl 7.2 2020-12-30 r5155 [x86_64-linux-5.10.176] (localbuild)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in STANDBY mode, exit(2)
+ date -u
Wed May 17 00:18:32 UTC 2023
+ lsblk -n -J -do NAME,TRAN,ROTA,RM,VENDOR,MODEL /dev/sdb
{
   "blockdevices": [
      {
         "name": "sdb",
         "tran": "usb",
         "rota": true,
         "rm": false,
         "vendor": "ASMT    ",
         "model": "CRCM535U31CIS   "
      }
   ]
}
+ date -u
Wed May 17 00:18:36 UTC 2023
+ smartctl -n standby -i /dev/sdb
smartctl 7.2 2020-12-30 r5155 [x86_64-linux-5.10.176] (localbuild)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Western Digital Caviar Blue Serial ATA
Device Model:     WDC WD1600AAJS-22PSA0
Serial Number:    WD-WCAP96933962
LU WWN Device Id: 5 0014ee 1560b29cd
Firmware Version: 05.06H05
User Capacity:    160,041,885,696 bytes [160 GB]
Sector Size:      512 bytes logical/physical
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ATA/ATAPI-7 (minor revision not indicated)
Local Time is:    Wed May 17 08:18:36 2023 CST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
Power mode is:    ACTIVE or IDLE

+ sleep 6s
+ date -u
Wed May 17 00:18:42 UTC 2023
+ smartctl -n standby -i /dev/sdb
smartctl 7.2 2020-12-30 r5155 [x86_64-linux-5.10.176] (localbuild)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in STANDBY mode, exit(2)
+ smartctl --json -n standby -i /dev/sdb
{
  "json_format_version": [
    1,
    0
  ],
  "smartctl": {
    "version": [
      7,
      2
    ],
    "svn_revision": "5155",
    "platform_info": "x86_64-linux-5.10.176",
    "build_info": "(localbuild)",
    "argv": [
      "smartctl",
      "--json",
      "-n",
      "standby",
      "-i",
      "/dev/sdb"
    ],
    "messages": [
      {
        "string": "Device is in STANDBY mode, exit(2)",
        "severity": "information"
      }
    ],
    "exit_status": 2
  },
  "device": {
    "name": "/dev/sdb",
    "info_name": "/dev/sdb [SAT]",
    "type": "sat",
    "protocol": "ATA"
  }
}
+ date -u
Wed May 17 00:18:43 UTC 2023
+ smartctl -n standby -i /dev/sdb
smartctl 7.2 2020-12-30 r5155 [x86_64-linux-5.10.176] (localbuild)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

Device is in STANDBY mode, exit(2)
root@OpenWrt:~#
root@OpenWrt:~#

```

chunfengyao, 2023-05-17:
Furthermore, based on the device information obtained from both `lsblk` and `smartctl`, it can be observed that lsblk only retrieved information pertaining to the HDD cage. In actuality, the information obtained by `smartctl` was actually from the HDD itself.

chunfengyao, 2023-05-17:
Working with smartctl:
`opkg install smartmontools smartmontools-drivedb`

chunfengyao, 2023-05-17:
> Furthermore, based on the device information obtained from `lsblk`, it can be observed that lsblk only retrieved information pertaining to the HDD cage. In actuality, the information obtained by `smartctl --json -i /dev/sdb` was actually from the HDD itself.

Command `smartctl --json -i /dev/sdb` at here will spinning up the hdd as well.
Is there any way to get the hot-plug event of hdd, so that we could retrieve and cache the device information associated with the HDD when it is plugged in?

psiegl, 2023-05-17:
Interesting. The idea to include the hard drive information was that for a - non technical - user, the hdd idle UI is close to not usable. How should he/she identify which hard drive will be put to sleep by just seeing sd*? As of this, I think it is significantly more convienient if you know at least the hard drive vendor and type. Sure, if there are multiple you have yet another issue.

I am aware about smartctl, but there is an issue with it. Not by the tool, but: if the hd-idle UI would employ smartctl, one would have to add a dependency to smartmontools in the opkg. Not sure, if the maintainer would like that. blkid in this regard is more convienient, as it can be found on any Unix system, especially also on OpenWRT.

So question is: how can one provide more information to a non-technical person about the hardware characteristics without querying the disks right? Because otherwise, they start waking up.

psiegl, 2023-05-17:
Interesting. The idea to include the hard drive information was that for a - non technical - user, the hdd idle UI is close to not usable. How should he/she identify which hard drive will be put to sleep by just seeing sd*? As of this, I think it is significantly more convienient if you know at least the hard drive vendor and type. Sure, if there are multiple you have yet another issue.

I am aware about smartctl, but there is an issue with it. Not by the tool, but: if the hd-idle UI would employ smartctl, one would have to add a dependency to smartmontools in the opkg. Not sure, if the maintainer would like that. lsblk in this regard is more convienient, as it can be found on any Unix system, especially also on OpenWRT.

So question is: how can one provide more information to a non-technical person about the hardware characteristics without querying the disks right? Because otherwise, they start waking up.

chunfengyao, 2023-05-17:
> So question is: how can one provide more information to a non-technical person about the hardware characteristics without querying the disks right? Because otherwise, they start waking up.

perhaps, grab the disk info when It was plugged in the system(the disk will be active when pluged in). then store these info.

If we can observe the plugged-in event of disk, then we can grab its info for the first time.

jow-, 2023-05-30:
Is this really a problem? How often do you open this page and continuously refresh it?

chunfengyao, 2023-05-30:
It's still a problem now, but just occasionally.
I hope this issue could be fixed.

chunfengyao, 2023-05-30:
Maybe, we could simply revert the code to the previous copy?
actually, It's hard to find an ingenious way to handle this problem.

- from the side of the system, It should provide the latest info of the disk to the user. so It should access disk to query the latest info(but, It's hard to maintain a cache of these info, and make sure It's latest).
- from the side of disk, even the system just want to read first sector of disk, It should been spinned up as well.

jow-, 2023-05-30:
Yeah, but why is this a problem in practice? If you don't want your disks to spin up, simply don't visit the page? What reason is there to open the hd-idle configuration page if you do not intend to deal with hd-idle? Does it provide any kind of information you wouldn't otherwise get?

chunfengyao, 2023-05-31:
put the text below in `/rtc/hotplug.d/block/` as file `00-hd-idle-luci`
```shell
_log_path=/tmp/hd-idle.luci.txt

env |tee -a "${_log_path}"
echo "===============" |tee -a "${_log_path}"

echo "$*" |tee -a "${_log_path}"
echo "===============" |tee -a "${_log_path}"

blkid |tee -a "${_log_path}"
echo "===============" |tee -a "${_log_path}"
```

then plug in a new disk, and unplug it.
we will got this from `/tmp/hd-idle.luci.txt`:
```shell
DEVNAME=sda
USER=root
ACTION=add
SHLVL=1
HOME=/
SEQNUM=2326
MAJOR=8
HOTPLUG_TYPE=block
DEVPATH=/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2.3/2-2.3.3/2-2.3.3:1.0/host2/target2:0:0/2:0:0:0/block/sda
LOGNAME=root
DEVICENAME=sda
TERM=linux
SUBSYSTEM=block
BOOT_IMAGE=/boot/vmlinuz
PATH=/usr/sbin:/usr/bin:/sbin:/bin
MINOR=0
PWD=/
DEVTYPE=disk
===============
block
===============
/dev/nvme0n1p1: SEC_TYPE="msdos" LABEL_FATBOOT="kernel" LABEL="kernel" UUID="1234-ABCD" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="abeae232-5289-82a1-0e57-2f2fe899f401"
/dev/nvme0n1p2: TYPE="squashfs" PARTUUID="abeae232-5289-82a1-0e57-2f2fe899f402"
/dev/loop0: LABEL="rootfs_data" UUID="9836005f-1d04-429c-a9da-7359a03d1323" BLOCK_SIZE="4096" TYPE="f2fs"
/dev/nvme0n1p128: PARTUUID="abeae232-5289-82a1-0e57-2f2fe899f480"
/dev/sda1: SEC_TYPE="msdos" LABEL_FATBOOT="kernel" LABEL="kernel" UUID="C9C0-FDDD" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="EFI System Partition" PARTUUID="529e51f2-6479-dfa3-f1d0-7fb0bec60d01"
===============
DEVNAME=sda1
USER=root
ACTION=add
SHLVL=1
HOME=/
SEQNUM=2327
PARTNAME=EFI System Partition
MAJOR=8
HOTPLUG_TYPE=block
DEVPATH=/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2.3/2-2.3.3/2-2.3.3:1.0/host2/target2:0:0/2:0:0:0/block/sda/sda1
LOGNAME=root
DEVICENAME=sda1
TERM=linux
SUBSYSTEM=block
BOOT_IMAGE=/boot/vmlinuz
PATH=/usr/sbin:/usr/bin:/sbin:/bin
MINOR=1
PARTN=1
PWD=/
DEVTYPE=partition
===============
block
===============
/dev/nvme0n1p1: SEC_TYPE="msdos" LABEL_FATBOOT="kernel" LABEL="kernel" UUID="1234-ABCD" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="abeae232-5289-82a1-0e57-2f2fe899f401"
/dev/nvme0n1p2: TYPE="squashfs" PARTUUID="abeae232-5289-82a1-0e57-2f2fe899f402"
/dev/loop0: LABEL="rootfs_data" UUID="9836005f-1d04-429c-a9da-7359a03d1323" BLOCK_SIZE="4096" TYPE="f2fs"
[... 156 lines cut ...]
BOOT_IMAGE=/boot/vmlinuz
PATH=/usr/sbin:/usr/bin:/sbin:/bin
MINOR=0
PWD=/
DEVTYPE=disk
===============
block
===============
/dev/nvme0n1p1: SEC_TYPE="msdos" LABEL_FATBOOT="kernel" LABEL="kernel" UUID="1234-ABCD" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="abeae232-5289-82a1-0e57-2f2fe899f401"
/dev/nvme0n1p2: TYPE="squashfs" PARTUUID="abeae232-5289-82a1-0e57-2f2fe899f402"
/dev/loop0: LABEL="rootfs_data" UUID="9836005f-1d04-429c-a9da-7359a03d1323" BLOCK_SIZE="4096" TYPE="f2fs"
/dev/nvme0n1p128: PARTUUID="abeae232-5289-82a1-0e57-2f2fe899f480"
===============
```

It would be friendly to display the label info obtained from blkid(or other utilities) to the users, the partition's label just like the disk name in Windows(for example "Local Disk").
Additionally, it is necessary to include conditions to verify if the disk has a label, and then we can execute the command below to get all the labels and combine them into a string of partitions in this disk:
```
blkid /dev/${DEVNAME}*|grep -v '/dev/nvme0n1:'| sed 's/ /\n/g'|grep 'LABEL='|awk -F'"' '{print $2}' |sed 's/\n/,/g'
```
when
`DEVTYPE=disk`

we can store this combined string in a path, such as `/var/run/${DEVNAME}/labels-hd-idle-luci.txt`
then just display it on the luci page.

chunfengyao, 2023-05-31:
grab partion info when `ACTION=add`, It means that the disk was just plugged in.

chunfengyao, 2023-05-31:
> Interesting. The idea to include the hard drive information was that for a - non technical - user, the hdd idle UI is close to not usable. How should he/she identify which hard drive will be put to [...]

partition label may be a more friendly info than vendor name.