openwrt/luci #6408: luci-mod-status: Local Interface channels are not displayed on the graph

dvlom99, 2023-05-31:
## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Status → Channel Analyzis -> radioX(5Ghz) or radioX(2.4Ghz)
2. wait for the scan to finish

## Actual behavior:
Local Interface channels not displayed on the graph:
[image]
[image]

## Expected behavior:
Local Interface channels are correctly shown on the graph

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.5'
DISTRIB_REVISION='r20134-5f15225c1e'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.5 r20134-5f15225c1e'
DISTRIB_TAINTS=''
```

jow-, 2023-06-07:
@Ansuel

Ansuel, 2023-06-07:
@jow- i just notice that this is on 22.03 and this is fixed in master

Required change to be backported https://github.com/openwrt/luci/commit/307e385ba19fff36e335d048c45fc8f667129bd8

jow-, 2023-06-07:
Fixed by 1c26abc138

closed 2023-06-07 by jow-