openwrt/luci #6409: [openwrt-23.05]luci-app-upnp:"There are no active redirects",but miniupnpd works fine.

wtfr-dot, 2023-05-31:
I built the latest OpenWrt-23.05, which includes dual WAN and Passwall support. I disabled nftables and compiled fw3 because mwan3 is incompatible with fw4 and Passwall. The debug logs of miniupnpd and the firewall both indicated that they were working fine. However, UPnP did not display any entries in the Luci interface. The same configuration worked perfectly in OpenWrt-22.03.

OpenWrt version information from system `/etc/openwrt_release`

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05-SNAPSHOT'
DISTRIB_REVISION='r23058-e2e7dd4450'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05-SNAPSHOT r23058-e2e7dd4450'
DISTRIB_TAINTS='no-all mklibs'
[image]

jow-, 2023-06-02:
Please provide the output of `iptables --line-numbers -t nat -xnvL MINIUPNPD` when it is working but not displayed in the gui.

wtfr-dot, 2023-06-02:
> Please provide the output of `iptables --line-numbers -t nat -xnvL MINIUPNPD` when it is working but not displayed in the gui.

Thanks for your support.

root@OpenWrt:~# iptables --line-numbers -t nat -xnvL MINIUPNPD
Chain MINIUPNPD (1 references)
num      pkts      bytes target     prot opt in     out     source               destination         
1           0        0 DNAT       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:54297 DNAT [unsupported revision]
2           0        0 DNAT       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:54303 DNAT [unsupported revision]
3           0        0 DNAT       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:54299 DNAT [unsupported revision]
4           1       56 DNAT       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            udp dpt:39416 DNAT [unsupported revision]

jow-, 2023-06-07:
It's a problem with the iptables package, in particular the `[unsupported revision]` part. It should show destination address and port there. Without that information, the UI can't work.

Nothing we can fix on the LuCI side. It's likely a bug in the iptables package. Which one is installed? `iptables-nft` or  `iptables-legacy` ?