openwrt/luci #6411: luci-proto-ipv6: ds-lite/6to4/6in4/6rd not recognized as valid protocols

chris5560, 2023-06-02:
## Steps to reproduce:

1. install ds-lite/6to4/6in4/6rd
2. goto Network->Interaces->Add new Interface->this protocols not offered

or add i.e. 6to4 as protocol inside /etc/config/network
then Network->Interfaces show:
Unsupported protocol type.
Install protocol extensions...

## Additional Information:

Model: Netgear WNDR4300
Architecture: Atheros AR9344 rev 2
Target Platform: ath79/nand
Firmware Version
  OpenWrt 22.03.5 r20134-5f15225c1e
  LuCI openwrt-22.03 branch git-23.119.80898-65ef406

jow-, 2023-06-02:
You need to install `luci-proto-ipv6`.

closed 2023-06-02 by jow-

chris5560, 2023-06-02:
It's not a problem of installed packages.
After installing these protocols you need to restart the router, so that web-server cache is cleared.