openwrt/luci #6413: LuCI should warn about upgrading packages

kannes, 2023-06-03:
## Steps to reproduce:
1. go to: System → Software → Updates

## Actual behavior:
1. Mr. McSmartypants sees huge list of available updates
2. Mr. McSmartypants knows to keep network devices up to date
3. Mr. McSmartypants decides to upgrade all those packages
4. Mr. McSmartypants wonders why there is no "upgrade all" button and scours the web
5. Mr. McSmartypants finds issues like https://github.com/openwrt/luci/issues/4602#issuecomment-729783278
6. Mr. McSmartypants finds https://openwrt.org/meta/infobox/upgrade_packages_warning
7. Mr. McSmartypants needs to reinstall/restore to make sure his router did not get screwed up :(

## Expected behavior:
The page could have a informative message saying what the wiki page says ("Upgrading packages may cause serious problems, including soft-bricking your device!"), maybe with a link to the wiki page https://openwrt.org/meta/infobox/upgrade_packages_warning to prevent dutiful users not to do obvious mistakes.

## Title
I am sorry but I do not know which package this concerns, I will update the title when someone tells me :)

systemcrash, 2024-05-01:
Closed via https://github.com/openwrt/luci/commit/4d23adc8d938214475f9c68b3c4c7c5a55c29743

closed 2024-05-01 by systemcrash

kannes, 2024-05-02:
Awesome, thank you so much! This will make LuCI much safer to use for newbies.