openwrt/luci #6421: Can't choose to IPv6 in diagnostics.

Karmylr, 2023-06-07:
### Describe the bug

When I clicked the drop-down arrow next to the `IPv4 Ping` button, only a horizontal line appeared

### OpenWrt version

r23040-ec6bbc5255

### OpenWrt target/subtarget

x86/64

### Device

J4125,VMware ESXi

### Image kind

Self-built image

### Steps to reproduce

Go to diagnostics and try to choose to IPv6 Ping.

### Actual behaviour

A line appears

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-06-07:
Duplicate of #6379

closed 2023-06-07 by jow-

Karmylr, 2023-06-09:
After git pull and rebuild, there is still this problem.

Karmylr, 2023-06-11:
@jow- After git pull and rebuild, there is still this problem.