openwrt/luci #6422: luci-base: menu doesn't regenerate on UCI change

wryun, 2023-06-08:
## Steps to reproduce:

1. Add a uci dependency to a JSON menu item (e.g. `"depends": {"uci": {"system": {"foo": 1}}`)
2. Modify that configuration option
3. Reload the page

## Actual behavior:

The menu won't change unless you logout (to clear the session) and remove /tmp/luci-indexcache*

## Expected behavior:

When saving UCI items from the frontend, the frontend should automatically regenerate the menu.

## Additional Information:

Custom build of 22.03.5.

jow-, 2023-06-08:
Not sure we really want to flush the menu *on every single* uci change. The menu depends mechanism was not really meant for that level of dynamic usage. Primary use case was masking features (such as hiding wifi or switch config if the relevant sections are not present).

wryun, 2023-06-08:
I looked into this because we were using it to show log file pages only in certain situations (i.e. when particular options are enabled).

I'd argue that this makes it behave in a less surprising way, the cost of rebuilding the menu is not that high and that uci confirm is already a somewhat slow process. But I can understand that it's less than elegant given that most of the time it wouldn't change.

Alternatively, if the idea is that it should really only be for entire sections, perhaps the individual options dependency support could be removed. Do you see a use case for this when from a user perspective the device has to be rebooted for changes to 'take'?

jjm2473, 2023-08-21:
@jow- I have notice that, too. After wifi module auto config, menu not shown "Wireless". Same issue appears on Docker, if I open luci on browser on boot, Docker maybe not start on that time, menu will not show Docker menu even after docker started.