openwrt/luci #6446: luci-app-openvpn: Unhandled exception during request dispatching

henok400, 2023-06-27:
Trying to install luci-app-openvpn 
openwrt luci totally crash give this error cant open router admin
Unhandled exception during request dispatching
`
error loading module 'ubus' from file '/usr/lib/lua/ubus.so':
	Error loading shared library libubox.so.20230523: No such file or directory (needed by /usr/lib/lua/ubus.so)

In [anonymous function](), file [C]
  called from function require ([C])
  called from function [anonymous function] (/usr/lib/lua/luci/util.lua:15)
  called from function require ([C])
  called from function [anonymous function] (/usr/lib/lua/luci/template.lua:4)
  called from function require ([C])
  called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:5)
  called from function [anonymous function] ([C])

In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 55, byte 48:
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:145:26)
  called from function build_pagetree (/usr/share/ucode/luci/dispatcher.uc:382:73)
  called from function menu_json (/usr/share/ucode/luci/dispatcher.uc:439:26)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:893:24)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `                this.L.invoke('require', 'luci.ucodebridge');`
  Near here -------------------------------------------------^
`

feckert, 2023-06-28:
Have you checked whether the libraries mentioned are installed on your system?

henok400, 2023-06-28:
i try to install openvpn that i follow step like previous openwrt version. 
Openvpn sucessfully install openwrt v19.** only other version not working or luci crash.
current version openwrt v23.05 try to install
```
opkg update
opkg install openvpn-openssl
opkg install luci-app-openvpn
```
then the last `luci-app-openvpn` forced install that router crash and admin cant open. 
this bugs of last version openwrt v23.05 
i restore openwrt v19.10 then install everything openvpn working perfectly. 
i reporting current version under bugs of install openvpn.

feckert, 2023-06-28:
> Trying to install luci-app-openvpn openwrt luci totally crash give this error cant open router admin Unhandled exception during request dispatching ` error loading module 'ubus' from file [...]

I think the ABI version of libubox.so.* is wrong on your system that ubus.so needs. Did you checked that?
What es result of the following command?
` ls -la /lib | grep ubox`

Draggon, 2023-06-28:
Hey
Same here on 23.05.0-rc1 (r23069-e2701e0f33)>
Installed version of libubox is 20220927.

jow-, 2023-06-29:
That's because the libubox package was updated to an ABI-incompatible version upstream. You need to install the -rc2 image to solve this issue. Nothing we can do on the LuCI side.

closed 2023-06-29 by jow-