openwrt/luci #6449: luci-base: Inaccessible via Safari when using TLS Client Certs

gongloo, 2023-06-29:
## Steps to reproduce:
1. Implement client certs on HTTPS server in front of Luci, either reverse proxy or otherwise.
2. Attempt to use Luci

## Actual behavior:
Safari on Mac repeatedly shows a client certificate selection dialog and repeatedly reloads the landing page.
Safari on iOS presents an error showing that no appropriate client certificate is available.
Chrome on Mac works as expected, showing the luci login page.

## Expected behavior:
Safari shows the luci login page.

## Additional Information:
Context: https://forum.openwrt.org/t/tls-client-certificate-luci-login-issues-on-safari/164024

> From what I can tell, Safari handles a 403 (Forbidden) response as pertaining to the client certificate, and thus tries to refresh the page after showing a client cert selection dialog to the user (on Mac). Luci's login page is displayed using a 403 response code.

Would it be reasonable to display the login page using a 200 response code instead? Happy to submit pull requests if I can (a) get guidance on what the recommended approach is here and (b) sufficiently wrap my head around the luci codebase.

jow-, 2023-08-03:
I'm reluctant to switch to a 200 code just to work around this fringe issue. Is there no other way? Some proprietary http reply header maybe?

gongloo, 2023-08-04:
Not that I'm aware of... Are there UX downsides to using a 200?

jow-, 2023-08-12:
Well any unauthenticated request to a URL that requires authentication will yield a 403. Simply switching to 200 would also make non-login API endpoints return 200 for unauthenticated requests, which would break expectations. Returning 200 on 403 for login pages only might work but it introduces ugly inconsistencies.

If at all, returning 403-as-200 should only be done if
 - the user agent is found to be Safari
 - the requested page URL has HTML login functionality enabled
 - the `HTTPS` environment variable is set to `on`