openwrt/luci #6467: Vulnerability Report: OpenWrt - Overview - LuCI 404 Page Reflected XSS Vulnerability

Hebing123, 2023-07-15:
Vulnerability Summary:
The LuCI web interface management tool within OpenWrt is affected by a reflected cross-site scripting (XSS) vulnerability. This vulnerability can be exploited by crafting specific malicious requests that allow an attacker to execute arbitrary script code on the affected 404 error page.

Vulnerability Description:
The vulnerability exists in the 404 error page of OpenWrt's LuCI web interface management tool. When accessing a non-existent URL, OpenWrt displays a 404 error page to indicate that the requested resource was not found. However, LuCI lacks proper input validation when handling this error page, enabling an attacker to inject and reflect malicious script code back to the user's browser.

Exploitation of the Vulnerability:
An attacker can construct a URL containing malicious script code and then send it to a targeted user, enticing them to click on the URL. When the user accesses the URL, LuCI's 404 error page reflects the malicious script code back to the user's browser, where it is executed. This could allow the attacker to perform arbitrary actions, such as stealing sensitive information or session hijacking.

Exploitation Example:
Here is a malicious URL example that triggers the reflected XSS vulnerability:
http://110.183.106.109:8089/cgi-bin/luci/admin/;%3B<%2Fscript><script>alert%289232%29<%2Fscript>
[image]
[image]

Recommended Solution:
To address this vulnerability, the OpenWrt project development team is advised to take the following security measures:
1. Implement strict input validation and filtering for user input, particularly when reflecting user input on error pages.
2. Properly encode and escape script code within user input to prevent the execution of malicious scripts.
3. Implement secure programming practices and adhere to best practices, including using up-to-date web development frameworks and libraries to reduce the likelihood of security vulnerabilities.
4. Conduct regular security audits and vulnerability scans to promptly identify and address potential security flaws.
5. Publish vulnerability patch fixes to the OpenWrt user community and advise users to.

jow-, 2023-07-15:
Fixed in https://github.com/openwrt/luci/commit/24d7da2416b9ab246825c33c213fe939a89b369c (master), https://github.com/openwrt/luci/commit/cd8bea94e61fa72a0a2ba7bc54d7b2d7b7572519 (openwrt-22.03), https://github.com/openwrt/luci/commit/5f30aa0e19edc5952ac8e74707597eb095ebc714 (openwrt-21.02)

closed 2023-07-15 by jow-

stangri, 2023-07-15:
@jow- what about 23.05?

hnyman, 2023-07-15:
24d7da2 was committed into master in March, well before 23.05 branching in May.
So, automatically taken from master to the new branch two months later...